vhelin/wla-dx #27: Don't use .DB in .STRUCT + workaround

cr1901, 2015-01-30:
The manual states that except for nesting .STRUCTs, anything that can be placed within an .ENUM is also valid in a .STRUCT. However, it appears .DB will not work properly, and will cause all offsets to become a random 32-bit integer. The behavior can be shown in any file by adding the following lines (using a WDC 6551 I/O map as an example):

```
.MACRO PRINTCR
	.printv HEX \1
	.PRINTT "\n"
.ENDM

.STRUCT ACIA_REGS
TDR .DB 0
RDR DB
RST .DB 0
STAT DB
CMD DB
CTRL DB
.ENDST

.ENUM $8000
ACIA INSTANCE OF ACIA_REGS
.ENDE

PRINTCR ACIA_REGS.TDR
PRINTCR ACIA.TDR
PRINTCR ACIA_REGS.RDR
PRINTCR ACIA.RDR
PRINTCR ACIA_REGS.STAT
PRINTCR ACIA.STAT
```
This .STRUCT/.ENUM is appropriate b/c TDR (TX) and RDR (RX) are write-only and read-only registers at the same address. Here, ACIA_REGS should be relative to $0, and ACIA should be relative to $8000. I get a random 32-bit integer in either case.

A workaround for the time being is to replace all instances of .DB with DSB 0:
```
.MACRO PRINTCR
	.printv HEX \1
	.PRINTT "\n"
.ENDM

.STRUCT ACIA_REGS
TDR DSB 0
RDR DB
RST DSB 0
STAT DB
CMD DB
CTRL DB
.ENDST

.ENUM $8000
ACIA INSTANCE OF ACIA_REGS
.ENDE

PRINTCR ACIA_REGS.TDR
PRINTCR ACIA.TDR
PRINTCR ACIA_REGS.RDR
PRINTCR ACIA.RDR
PRINTCR ACIA_REGS.STAT
PRINTCR ACIA.STAT
```

Ville: Incidentally, this may be related to the bug I found back in 2013 that spurred me to originally get in contact with you. Since I have sample code to test, I'll assign myself to the bug and see what I can find (when I have a moment- I'm actually working on a ROMable piece of code using WLA :P).

vhelin, 2015-01-30:
> The manual states that except for nesting .STRUCTs, anything that can be
> placed within an .ENUM is also valid in a .STRUCT. However, it appears .DB
> will not work properly, and will cause all offsets [...]

Oh my - what a bug. If you don't find anything, let me know and I'll give
it a look.

- Ville

cr1901, 2016-06-03:
Re: this bug...

I can't find any evidence that either .ENUM or .STRUCT have any logic to handle the dot cases since the code was placed into git version control. 

The bug changed (presumably due to C UB handling) so that my example (the one with .DB) errors out while trying to parse RDR, thinking that's it's parsing a bogus .STRUCT directive instead of a struct field.

I set a breakpoint at line 2304 in pass_1.c, and this is what I found:

```
$11 = {name = "TDR\000\rd-º\rd-º\rd-º\rd-º\rd-º\rd-º\rd-º\rd-º\rd-º\rd-º\rd-º\rd-º\rd-º\rd-º\rd-º", size = -1163005939, next = 0x962210}
(gdb) p *st->items->next
$12 = {name = "0\000-º\rd-º\rd-º\rd-º\rd-º\rd-º\rd-º\rd-º\rd-º\rd-º\rd-º\rd-º\rd-º\rd-º\rd-º\rd-º", size = -1163005939, next = 0x0}
(gdb)
```

Notice that size for both of them is invalid, and in the case of RDR, the name that was parsed was a null string. I am not currently sure why WLA didn't try to parse DB (without the dot) as the next struct field, but I'm fairly confident this is the culprit: https://github.com/vhelin/wla-dx/blob/master/pass_1.c#L2301

nicklausw, 2016-06-03:
This doesn't really help anything since I think the source from this point in time is lost, but: http://www.network54.com/Forum/56397/message/978526711/Wow+thanks%21

vhelin, 2017-05-13:
Thanks to Maxim for the fix to this issue, it's now in the sources.

It seems that struct definition skips all that begins with a comma, so if you want a byte size item "test" inside a struct, you'll need to use db, not .db

.struct testStruct
test:  db
.endst

Alternative, like you said, you could use dsb for the same effect, e.g.

test: dsb 1

closed 2017-05-13 by vhelin

cr1901, 2017-05-13:
Begins with a comma? None of my test cases have commas.

And doesn't `DB` increment the struct pointer?

vhelin, 2017-05-14:
Sorry, begins with a dot I meant.

DB increments struct pointer, but .DB is just skipped without any effect.