vhelin/wla-dx #28: Support conditions inside enum

maxim-zhao, 2015-02-13:
e.g. when using an enum for RAM allocation:
```
.enum $c000
someVar db
.ifdef debug_enabled
someDebugVar db
.endif
.ende
```
I'd expect .if to work pretty much everywhere, much like the C preprocessor, although I know it isn't the same.

NicklausW, 2015-02-13:
I do think this should be a feature.

In the meantime, have you considered reversing it, and if debug_enabled is defined, make a separate .enum with all that debug data?

maxim-zhao, 2015-02-15:
That's a workaround, but an ugly one as I'd have to keep both enums in sync. My particular case is a disassembly, so I can't reorder things - my goal is to have "assemble to match the original" mode and "miss out unnecessary stuff" mode, where I can make use of the assembler features to omit unecessary things. I'll just leave that for a later stage once all the RAM is accounted for.

vhelin, 2015-02-16:
> I do think this should be a feature.

I think so, too. Next time I'll work on WLA I'll investigate how big
changes making this work would require.

nicklausw, 2016-03-29:
An idea I had a few months ago is putting the .IF directives in their own function. The main directive check function can just call it and return if it succeeds, and directives like .ENUM that contain their own little parser can use it too.

nicklausw, 2016-03-29:
Personally, I don't like how the assembler parses code too much (it would work well if some directives didn't have their own parsers). Not really sure how I'd make it better.

Drenn1, 2018-04-16:
This can be closed, it was addressed with #150.