vhelin/wla-dx #29: UTF-8 support

maxim-zhao, 2015-02-16:
It's sometimes restrictive to have to use ASCII, e.g. not able to use extended box drawing characters. I guess people using accented characters would find it useful too. I can imagine this would be a very painful change, though...

```
.asciitable
map '£' = 39
.enda
```
```
INPUT_NUMBER: Got 'ú' (163) when expected "'".
ERROR: Couldn't parse "map".
```

vhelin, 2015-02-16:
Oh man, making this work would require changes all over the code. :) Linker
less, assembler a lot more...

cr1901, 2015-02-16:
I don't think that can be done without making the code non-ANSI C, (which admittedly is my goal to keep the code ANSI-fied, not Ville's). Personally, I think the gain is too little for the work that is required.

If I need non-ASCII characters, I'd just make a define that describe the character:
```
.DEFINE pound_char 0x39
```

If you're meant to print the character on the target system, you'll need to create your own set of glyphs in a paint program anyway :D.

maxim-zhao, 2015-02-16:
Yes, I was just hoping for stuff I can embed in strings. Here's my use case:

https://github.com/maxim-zhao/smsgraphicboard/blob/8ca191f2c1a03abce88ada328c554f10a5b88ab8/Dasm/Sega%20Graphic%20Board%20v2_0%20%5BProto%5D.sms.asm#L2310

It's helpful for layout to have a single character inside the `.asc` map to a glyph.

Imagine using WLA to work with accented text, or Japanese - you really want to say
```
.asciitable
map 'ぁ' to 'ん' = 0
.enda
.asc "ひらがな"
```
Doing all that with definitions would be nightmarish - you'd probably go to .tbl files and third-party tools to preprocess the text.

cr1901, 2015-02-16:
Your changes would require WLA to be Unicode-aware wrt input source files. I'm not about to add a dependency on libiconv in the main branch, but maybe there can be a fork that is Unicode-aware.

In any case, this is something I have to think about/get back to you.

cr1901, 2015-11-15:
Been thinking about this some more; one of the bad things about only knowing/requiring English on a daily basis is that I'm really not qualified to implement i18n. 

Perhaps what should be done is to have a define for UNICODE support and ASCII support. Or perhaps a WLA mode for Unicode variable-length lexer/parser, and ASCII lexer/parser (and again, just make sure that Windows users save as UTF-8).

I can see this become hellish very quickly. However, this is an important change; care must be taken to ensure it doesn't break ANSI C compat however.

maxim-zhao, 2022-03-06:
I've been playing with this a bit more, with the mostly theoretical use case of emoji in code 💩✨ which I think acts as a decent way to throw the worst case scenario at you. UTF-8 should mostly "just work" in most contexts because it's just bytes >127; only when WLA DX tries to be restrictive do we have trouble.

Here's some things that seem to work:

1. Labels

```
    call 💩

💩:
  ret
```

2. Comments - I'm sure people have been using accented characters in comments for years. 
3. Strings in general

```
.section "🎉" free
.print "💩\n"
.db "✨"
```

4. Macro names (except for the first letter)

```
.macro X💩
  nop
.endm

  X💩
```

Here's some things that don't:

1. Definition names

```
.define 💩 1
```
```
DIRECTIVE_ERROR: A computation cannot be output as a string.
```

2. Labels when there's no whitespace between the `:` and an opcode:

```
    call 💩

💩:ret
```
```
ERROR: Couldn't parse "💩:ret".
```
Is there some different code path for this case?

3. First letter of a macro

```
.macro 💩
  nop
.endm

  💩
```
```
LOGIC_ERROR: Invalid value
```

vhelin, 2022-03-06:
I guess there are some places in the code that checks that the input is a lowercase char, uppercase char, number, underscore plus some other symbols and accepts only that. We might need to change that code to check that the input is not -+*/%\n etc and accept the rest...

vhelin, 2022-03-06:
Gotta say I find code like this so funny that it really makes me want add UTF-8 support to WLA :)

```

.macro 💩
  nop
.endm

  💩

```

maxim-zhao, 2022-03-07:
There's a certain elegance to it :) 

Another one for the list: you can't use extended characters as the first character of a label without a ':' suffix:

```
  call 💩

💩     ret
```
```
ERROR: Couldn't parse "💩".
```

I've come across `get_next_plain_string()` which has a whitelist for acceptable characters; but it also does something weird with curly braces `{}` which causes it to accept all characters until the braces are balanced. Can you explain why? Was it added for the "Substitution" feature? How does `get_next_plain_string()` differ from `get_next_token()`? The former seems to be used in places expecting a "name", whereas the latter maybe has to expect value-ish tokens?

maxim-zhao, 2022-03-07:
...and changing it to accept chars >127 (or <0 as it's treated as signed here) doesn't seem to fix the issues I've seen.

vhelin, 2022-03-07:
Yeah, {} handling was added for substitution...

Gotta say I don't remember about the rest, but I'll try to give this a closer look later this week.

maxim-zhao, 2022-03-07:
I found a fix for some of the above, PR coming...

maxim-zhao, 2022-03-07:
In fact I think that covers all of the issues (apart from `.asciitable` I guess which is worked around with `.stringmaptable`). Now you can be fully emoji compliant :)

vhelin, 2022-03-07:
Holy crap your solution is simple! :) Needs a little bit of polish, but I'm loving it...

cr1901, 2022-03-08:
Maybe there should be a Unicode test, especially for Amiga and DOS (since I suspect those platforms are going to break due to this PR :P).

vhelin, 2022-03-08:
Yeah, we'll definitely need a Unicode test!

vhelin, 2022-03-08:
I'll try to do that Unicode test later today, after work...

vhelin, 2022-03-08:
Forgot to add reference to this issue in the commit message, but there is now a UTF-8 test in WLA DX, under tests/z80. Azure Pipelines CI found no issues with that, but yeah, it might fail on Amiga/MSDOS or at least print weird characters on screen.

vhelin, 2022-03-08:
My UTF-8 test works on WinUAE, but yeah, printing those UTF-8 symbols to console just outputs garbage... :)

[image]

cr1901, 2022-03-08:
Well, the test passes, good enough for me. Being able to read UTF-8 is more important than being able to print it on these old platforms.

Perhaps in the future there could be an option to read Amiga/DOS charsets (I'm guessing both use chars 128-255 and are fixed width?) from WLA source files if it's _really_ needed.

maxim-zhao, 2022-03-09:
Technically you also get garbage printed in Windows unless you set the code page to UTF-8, and by default it's unlikely to be handled by the font; but the data is intact.

maxim-zhao, 2022-03-09:
Also this is just making it encoding agnostic - chars > 127 are just passed through. Thus almost any encoding will pass through, except for some of the legacy MBCS ones which use <127 bytes as part of a multi-byte sequence.

Also also: this was only not working due to two things: first, code acting like char is unsigned when it is presumably platform dependent. That's really a flaw in the code; I guess you ought to explicitly change between "texty" and "numbery" interpretations. Second, a few places in the code use a whitelist to capture a "word" that isn't an expression, rather than blacklist expression characters. That's down to the parser being as-hoc rather than grammar-based, but we've seen it's hard to change that.

maxim-zhao, 2022-05-15:
I think this is done...

closed 2022-05-15 by maxim-zhao