vhelin/wla-dx #30: Can't use .instanceof in enum

maxim-zhao, 2015-02-19:
Docs say:
> DB, DW, DS, DSB, DSW and INSTANCEOF can also be in lowercase. You
can also use a dotted version of the symbols, but it doesn't advance
the memory address. Here's an exmple:

(Note, typo at the end)
But:
```
.struct XY
  y db
  x db
.endst
.enum 0
  foo .instanceof XY
  bar dsb 100
.ende
```
```
DIRECTIVE_ERROR: Unexpected symbol "bar" in .ENUM.
ERROR: Couldn't parse "bar".
```
In fact, anything that comes after the struct name is rejected.

----

Maybe there's a better way to achieve what I want to do here? Namely:

1. I have Z80 code using the index registers, so it's super useful to use the struct member offset definitions (e.g. XY.x above) for the relative distances. Thus I want to use structs.
2. I also have cases where two totally separate parts of the code (title screen and in-game) want to make use of the same RAM locations for unrelated purposes. 

If I have .instanceof then I can make suitable structs to overlay the same part of RAM (much like a C union of structs) and put them in the same containing .enum. If I don't then I guess my workaround is to split my .enum up, so I can have two enumerating through the relevant bytes, but then I'm required to manage the start addresses to avoid unintended clashes - which is the problem .enum solves in the first place.

NicklausW, 2015-02-19:
Wait, what happens if you use instanceof instead of .instanceof?

maxim-zhao, 2015-02-19:
Well, it works - my point is, I want it to overlap with the "bar" array. How about a better example...
```
.struct InGameData
  foo db
  bar dsb 20
  baz dw
.ends
.struct TitleScreenData
  buffer dsb 16
  counter dw
.ends
.enum $c000
  ; Some items before the overlapped bit. Let's assume they're used by all the code.
  var1 db
  var2 dw
  ; I want to put an instance of each struct here, overlapping
  RAM_TitleScreenData .instanceof TitleScreenData
  RAM_InGameData instanceof InGameData
  ; more stuff here
  var3 dw
.ende
```
Trying to define all the relevant items without structs would be tough due to the way they use non-coincident chunks of RAM, and I want a struct for one or more of them anyway. Having .instanceof still requires me to select the largest struct to advance the enum (if I want to guarantee that following items don't get overlapped).

I guess really, I'm wondering if there's a better way to use the same RAM in distinct phases of my program, and use .enum to generate the addresses for me, without having to use dedicated .enums for the reused memory.

it's also the case that the docs suggest .instanceof works, and it doesn't. If it isn;t expected to work, then this item can be closed by changing the docs :smile:

maxim-zhao, 2015-02-19:
It looks like all the dotted parametrised items don't work as intended...
```
.enum 0
  foo .dsb 8 ; makes next symbol "unexpected"
  bar .instanceof my_struct ; likewise
  baz .db ; OK
  qux .dw ; OK
  quux .dsb ; OK!
  corge .dsw ; OK!
  grault .instanceof ; OK!?!
  some_real_allocation dsb 16 ; just so this isn't an entirely contrived example
.ende
```

vhelin, 2015-02-19:
Ok, I'll definitely have to investigate what's wrong here. :)

vhelin, 2021-03-15:
Just revisiting old and neglected issues... :) Does .UNION (cannot find when it was added in CHANGELOG) fix this issue?

maxim-zhao, 2021-03-17:
I'll have to dig out the project and try it...

maxim-zhao, 2022-11-12:
Well, I should follow up on this. Yes, unions work fine here and are what I use for double-purposing RAM. Still, for the original bug report: should dotted entries work, and if not, they should be explicitly rejected and the docs updated (if you didn't do that already).

closed 2023-04-29 by vhelin