vhelin/wla-dx #31: Remove or change "debug" opcode from GB.

AntonioND, 2015-04-11:
https://github.com/vhelin/wla-dx/blob/master/opcodes_gb.c#L123

I know, i know, there's a comment that says that "note that this works only with wzonka-lad", but it encourages people to use bad debugging methods. Other emulators, like no$gmb and bgb use opcodes with no effect ("ld b,b") for debugging purposes since they are ignored by other emulators and hardware. Using an undefined opcode is bad practice, in a real gb (and a good gb emulator) it hangs the program so, if the coder forgets to remove even one "debug" opcode, it will crash. If the user forgot a "ld b,b" the program would still work fine everywhere.

vhelin, 2015-06-21:
Right, I removed the opcode. :)

closed 2015-06-21 by vhelin

Guernouille, 2015-06-21:
Indeed, ld b,b / ld d,d and so on is the best choice for debugging purposes.