vhelin/wla-dx #33: Local Labels

Drenn1, 2015-05-01:
First off I know that if you put an underscore before a label name, it exists only in the given section. I also know about nameless labels. However, I would find it extremely useful to be able to make a local label which will fall out of scope after the next non-local label.

Requesting this because I found myself jealous of the pokered disassembly, which uses rgbds and puts a .done label at the end of every function which needs it. The readability definitely beats unnamed labels.

Drenn1, 2015-10-17:
Okay so, I kinda-sorta dealt with this myself. I didn't touch wla because I didn't really want to be responsible for bugs and whatnot, plus I wasn't sure how I'd even do it. Instead I wrote a [hackalicious script](https://github.com/Drenn1/ages-disasm/blob/82205762be767148ac4c0923ea86cd3f3cc0be6e/tools/wlaParseLocalLabels.py) which runs through my assembly file. Labels starting with a dot are considered "ultra-local" labels. Sorry if this is unhelpful, but well, I'm the only one who asked for this feature anyway :P

Anyway it looks like this if you're interested:

```
function1:
    ; do stuff
.done:
    ret

function2:
    ; do different stuff
.done:
    ret
```

galaxyhaxz, 2015-10-20:
May I make a suggestion? If this feature gets added, make it work like the above but use "@" instead of ".", since commands use a period. Also, "@" was used in real life here's a sample from the B.O.B source code (excuse the bad formatting):

	lda	#1
	sta	gettitlemusic
	bra	@gameover

	@notademo:
	lda	lives
	bne	@notgameover
	lda	#1
	sta	gotocontinue
	bra	@gameover

	@notgameover:
	dec	lives
	inc	winflag2

	@gameover:
	stz	demoflag
	jmp	resetloop

Neui, 2015-10-24:
I would like to prefer to have a "option" with dot because it is faster *(for me)* to write. Maybe if it ends with a `:`, then act as the local lablel?

Example:

	.PRINTT "Now assembling function UnSign"
	UnSign: ; Removes sign in 16-bit A
		PHP
		REP #$40
		BIT #$8000
		BNE .done ; Jumps to UnSign.done
		EOR #$FFFE ; XOR it with FFFE, wich deletes/add the minus sign
	.done: ; Creates UnSign.done or so
		PLP
		RTL

*Ok, to be honest, that might not be a good example because a normal `+` would be enough)*

vhelin, 2015-10-24:
As long as you guys don't want to make your own fork / branch, WLA DX will
reserve that leading / prefix dot only for directives / commands. :)

closed 2017-07-05 by Drenn1