vhelin/wla-dx #34: Remove Need for ".b" and ".w"

nicklausw, 2015-05-28:
The way wla-6502 works, the assembler can't tell on its own whether or not certain instructions should be put in as with 8-bit or 16-bit arguments, and so takes a default of the smallest possible argument. This means that unless you end some instructions with ".w", an error could come up saying something isn't an 8-bit value when the instruction can take 16-bit values.

An example:
lda $2002
would make an error because WLA thinks it's supposed to be 8-bit.

Why is this how it works? Can't it be like other assemblers where it defaults which instruction to use based on the value of the arguments given?

vhelin, 2015-05-28:
Hmm, the ways WLA works is that it should first try the 8bit version of the
instruction, and if the data was more than that, then use the 16bit
version. And that really should be the default behavior, without any .b and
.w hints. If this is not the case, then something must be wrong - I'll need
to check it out.

It can be that WLA chooses the 8bit version when it assembles, if the data
size is unknown, and when the assembled code gets to the linker, the linker
will find out that the data is 16 bits... Anyway, let's see what I can find.

nicklausw, 2015-06-18:
Any update on this? I'm being honest, this bug is the only big reason I'm using an assembler other than WLA for 6502 development. (Note that the bug is also in wla-56816 and possibly spc700).

vhelin, 2015-06-21:
No updates so far, I've been trying to get a job, but I just felt like
contribbin time to WLA, and committed a patch - I guess I could take a look
at this tomorrow?

nicklausw, 2015-06-29:
Man, this is one proud day for myself. I got bored of waiting to see if you'd find what was going wrong, so I took a look myself and seem to have found a fix.

After line 266 of parse.c, I added something along the lines of:

if (d > 0xFF)
{
operand_hint = HINT_16BIT;
} else {
operand_hint = HINT_8BIT;
}

I'm not doing the whole merge pull request thingy for two reasons:
1. The code might not be all that good
2. You might want to make the code a bit more bugs-proof
3. I don't exactly know how.

But yeah, according to my hex editor, the tests I did without a '.w' on a 16-bit value did fine.

nicklausw, 2015-06-29:
Alright, I went through on my own and solved the problem by placing the following after line 283:

`if (d > 0xFF)
{
operand_hint = HINT_16BIT;
} else {
operand_hint = HINT_8BIT;
}`

There's probably a more elegant way to do this, but hey, it fixed the bug for me.

nicklausw, 2015-06-29:
(Double postin' cuz tired of edits, heheh).

I've found that the best thing to do about this situation is (along with my existing fix) just precede my code with the .16bit directive. That makes it so that you don't really have to worry about 'this isn't 8-bit!' at all. WLA doesn't seem to have a problem with the opposite, .16BIT with 8-bit args...

Currently looking to see if there's a place in the code where .8BIT is defaulted to. Might change that in my fork.

bazzinotti, 2015-08-26:
here's what happens when trying to link main.o compiled from wla-huc6280:
```
FIX_REFERENCES: Value ($2215) of "bg_start_tilenum" is too much to be a 8bit value.
```
this error references the line `sta bg_start_tilenum` which is a RAMSECTION label. So far I've had to manually append `.w` to all my word-size labels.

nicklausw, 2015-08-29:
Try placing ".16bit" at the start of your code, it fixes the labels (but is inefficient as it removes ability to use zero page since everything will obviously be 16 bit).

bazzinotti, 2015-08-30:
I know it's imperfect that's why I don't want to use it

nicklausw, 2015-09-29:
I've been thinking, and maybe the object format could have a new addition called pending opcodes, where there are two or three possible opcodes given depending on the end value of the parameter. This would require some funky parsing on the part of WLA, though, so funky that I of course can't figure out how to do it (have never been able to understand how WLA works with the opcode tables).

cr1901, 2015-11-15:
`lda $2000` really should work w/o directives (and I think at one time it did). Marking as both bug and enhancement until I have time to work on it.

nicklausw, 2015-11-15:
I'll have to test some older versions and see if the directives weren't necessary once. Who knows, maybe we could figure out what all happened that screwed it up.

nicklausw, 2015-12-30:
UPDATE: this is super late on my part, but I can confirm that WLA 7.8 can, in fact, handle defines without any ".b" or ".w".

My binaries of said version: http://www.mediafire.com/download/t0xrlpjqxcm3zvl/wla-7.8.zip
My highly stripped down test rom: http://www.mediafire.com/download/tj7j1r1gw06m6fb/wla-dx.s

7.8's sources can be found on Ville's site. I say my sources are highly stripped down because 7.8 doesn't have lowercase letters inside rombankmaps, anonymous labels, forced sections, orga...quite  the pain. And due to the lack of orga, the program doesn't assemble into what it's supposed to be anymore, but still does so properly other than that.

Note: pre-8.0 releases contain the linker inside wla-xxxx and wlalink, both of which work.

Also apparently Ville found his old WLA sources (1.6-4.4). I'm building them right now and playing around with their (lack of) functionality. Fun :+1:

vhelin, 2015-12-30:
> UPDATE: this is super late on my part, but I can confirm that WLA 7.8 can,
> in fact, handle defines without any ".b" or ".w".
>
> My binaries of said version:
> http://www.mediafire.com/download/t0xrlpjqxcm3zvl/wla-7.8.zip
> My highly stripped down test rom:
> http://www.mediafire.com/download/tj7j1r1gw06m6fb/wla-dx.s
>
> 7.8's sources can be found on Ville's site. I say my sources are highly
> stripped down because 7.8 doesn't have lowercase letters inside
> rombankmaps, anonymous labels, forced sections, orga...quite the pain. And
> due to the lack of orga, the program doesn't assemble into what it's
> supposed to be anymore, but still does so properly other than that.
>
> Note: pre-8.0 releases contain the linker inside wla-xxxx and wlalink,
> both of which work.
>
> Also apparently Ville found his old WLA sources (1.6-4.4). I'm building
> them right now and playing around with their (lack of) functionality. Fun [image:
> :+1:]

Have... fun! :D

Yeah, I managed to save the stuff on my oldest storage HDD (external,
Western Digital) by taking the HDD out from it's plastic casing. Apparently
the casing's glue logic had died as the drive said nothing when plugged
into a computer, but the HDD itself worked "ok" (got around 2MB/s out of
it). And there I found all those old sources, among other things... :)

BTW, I think I saw some other old WLA sources on some other storage drive,
gonna take a look soon.

But yeah, pre 8.0 WLA itself could do the linking, but I got bored of
having the same kind of code in two places and thus moved all that had to
do with linking/making the final package to WLALINK. Maintaining the code
got a lot easier ever since.

...

Anyway, sorry guys for the lack of attention, I've had a busy end of the
year, and I'm still busy with a client project and finding a new apartment
etc... :P I hope I can give some time to WLA DX soon...

nicklausw, 2016-02-18:
Can someone point out to me where in the code pending computations are actually made for opcode arguments? I can't even find a mention of it.

nicklausw, 2016-02-18:
And something that I never pointed out: I'm pretty sure `lda $2000` still works. Labels are what cause problems, and defines should be made before the actual code regardless.

nicklausw, 2016-04-16:
We don't need two separate issues for the same problem.

closed 2016-04-16 by nicklausw

vhelin, 2016-04-19:
This should be true; The assembler chooses the opcodes, and for "lda $2002"
it should be able to see that it needs to use the 16bit version, but "lda
WHATAMI" it has to guess if the assembler doesn't immediately know what
WHATAMI is as WHATAMI could be anything, could be inside some other file
that the linker links later thus "lda WHATAMI.w" is needed, if you want the
16bit version...

bazzinotti, 2016-04-19:
Why not have a "later pass" that is more prepared to make correct decisions about the opcodes?

vhelin, 2016-04-19:
And that should be then in the linker as it's too early to have that later
pass in the assembler... And in that case the assembler should feed the
linker just its intermediate (language) code / metadata stream, not just
raw data with holes (from pending computations).

Sounds like a partial rewrite. Want to do it? :D

bazzinotti, 2016-04-19:
Possibly I would be comfortable doing this **after** you teach me WLA DX
Architecture (please teach me at some point!)

nicklausw, 2016-04-19:
Ville, now may or may not be a good time to ask, but what does pass 3 of the assembler even do? There's no docs that seem to explain the internal passes very well, from what I've found.

bazzinotti, 2016-04-19:
Consider joining IRC EFnet #wla-dx if you all want to chat :)

vhelin, 2016-04-19:
About pass 3 etc., I'll write all I know to README, asap. Can't really IRC
over this 3G connection from this beach bar, disconnects all the time... :)

wrote: