vhelin/wla-dx #36: Inter-bank sections

Drenn1, 2015-07-17:
I wonder how difficult this would be to implement. Some kind of INTERBANK-type section which can cross over banks.

A bit about where I'm coming from: I'm working on a disassembly for Oracle of Ages using wla-dx, and there are several instances of data which can flow across banks without interruption. Currently I'm using a series of macros or, in one case, python scripts to handle this. The disadvantage to this approach is that I need to hard-code the start address of the data, since I need to do calculations with the address at compile time (or even before that when using python scripts). This isn't a huge problem, but being able to use sections for this would be ideal.

cr1901, 2015-07-18:
Would you be willing to create a minimal gist illustrating what you wish to do/what your current Python script is doing? Tbh, I'm having trouble visualizing what you mean.

Drenn1, 2015-07-18:
Here's a before and after kind of thing: https://gist.github.com/Drenn1/a8f2d7473eb16f254aea

The first byte of the file it incbins is "metadata" indicating a compression mode, just in case you're wondering why I do "SKIP 1" and such.

Anyway as you can see, the actual usage of the macro isn't all that different when using sections, but the big draw is that I could rely on wlalink to ensure that it can find free space, instead of hard-coding the address it starts at.

Don't feel pressured to add this, I just like to throw my thoughts and ideas around. I'm making do without it, this would just be... cleaner I guess.

vhelin, 2015-07-18:
> I wonder how difficult this would be to implement. Some kind of
> INTERBANK-type section which can cross over banks.
>
> A bit about where I'm coming from: I'm working on a disassembly for Oracle
> of Ages [...]

The linker would need to be modified so that it would split sections that
overflow from a bank to the next one, into two, and I'm not 100% sure this
moment how I'd handle e.g. 16bit pending computations where the 1st byte
would go into bank A, and the 2nd byte into bank A+1... Also, what would be
the slot for the code/data that goes into bank A+1?

Drenn1, 2015-07-18:
Hmm, this could be confusing to implement on a general scale. On the gameboy bank 1 and onwards would all use the same slot. But since a bank isn't really tied to a specific slot, how would you tell it which banks are okay to use...? Perhaps you could simply tell it banks X through Y are okay to use, and to use slot Z for all banks. That would work for me, anyway, but perhaps that wouldn't be friendly to certain other systems, I'm not sure.

closed 2016-01-15 by Drenn1