vhelin/wla-dx #37: Reserve checksum and complement check bytes

nicklausw, 2015-07-20:
In wla-gb, I have a very large section (2 kb, although that doesn't really relate to this,) and the beginning of it gets overwritten because it starts at where the checksum + complement check will be.

Looking in the source, there doesn't seem to be a good way to reserve these bytes without giving an absolute value which, while it fixes the problem, also gets overwritten and looks ugly.

Uh...any ideas?

vhelin, 2015-07-20:
My suggestion: if the user makes the linker to calculate checksum etc., the
linker should reserve those bytes itself before it places the sections.
I.e., a new feature for me to implement. :)

nicklausw, 2015-07-20:
Wouldn't it just be a case of reversing the order of placement of checksums and sections? (At least, for this case in specific).

vhelin, 2015-07-20:
Yeah well, one needs all the data placed into the output before one can
calculate the checksum. :)

nicklausw, 2015-07-20:
It seems I've forgotten how checksums work.

vhelin, 2015-08-01:
Hey, can you test the latest sources - I made the checksummers reserve the bytes, but don't have the time to test them thoroughly.

nicklausw, 2015-08-01:
Works. Thanks!

closed 2015-08-02 by vhelin