vhelin/wla-dx #41: .ALIGN directive

Drenn1, 2015-07-29:
An .ALIGN directive would be particularly useful for Ramsections, in my opinion. Particularly on the gameboy, you may have buffers which need to start at a CX00/DX00 address. I guess it could also be useful for data in general, though since sections can be aligned, it's not as necessary. I might actually see about implementing this myself, I'd just like to get an OK from one of the maintainers first.

vhelin, 2015-07-29:
> An .ALIGN directive would be particularly useful for Ramsections, in my
> opinion. Particularly on the gameboy, you may have buffers which need to
> start at a CX00/DX00 address. I guess it could also [...]

Ouch, .SECTIONs already have ALIGN, but .RAMSECTIONs don't... Might just be
a copypaste job? Anyway, you have my blessings for this if you want to
check it out. I might also find time for WLA stuff later this week if you
don't have the time...

Drenn1, 2015-07-29:
Okay, I'll see what I can do. While we're on the topic of ramsections, I have a question about their behaviour. If I have this:

```
.RAMSECTION "RAM 2" BANK 2 SLOT 2

w2Data: dsb $40

.ENDS

; Bank 3: tileset data
;
.RAMSECTION "RAM 3" BANK 3 SLOT 2

w3Data: dsb $40

.ENDS
```

Assuming SLOT 2 begins at $d000, then w2Data will point to 2:d000, however w3Data will point to 3:d040 instead of 3:d000. Is this intended behaviour, and if it is, how can I reset the position of the Ramsection? The workaround I'm using right now is to define multiple slots, and use a separate one for each bank, though that seems silly.

vhelin, 2015-07-29:
> Okay, I'll see what I can do. While we're on the topic of ramsections, I
> have a question about their behaviour. If I have this:
>
> .RAMSECTION "RAM 2" BANK 2 SLOT 2
>
> w2Data: dsb $40
>
> .ENDS
>
> ; Bank 3: [...]

I'm not 100% sure, but I think the .RAMSECTIONs fill just the slots, not
banks. I guess as I added them much later than .SECTIONs, they were quickly
done and I guess I just had systems in mind that didn't have ram bank
switching. Perhaps if we added a directive called .RAMBANKMAP (much like
.ROMBANKMAP) we'd get .RAMSECTIONs working properly?

Drenn1, 2015-07-30:
It may suffice to simply treat a ramsection with a given slot and bank independently from other ramsections with different slots or banks. I feel like defining a .RAMBANKMAP could be more complicated than a ROMBANKMAP, since there could be ram at different locations, each location with its own bank system. And I don't really see an advantage to that as opposed to the more simple solution.

vhelin, 2015-08-02:
True, true... I'll add separate issues/enhancement tasks about .RAMSECTION
ALIGN(ment) and the use of BANK in section placement, and close this one,
and try to implement these in the next couple of days.

closed 2015-08-02 by vhelin

cr1901, 2015-08-16:
I'd like to add: Bank references in .RAMSECTIONS work correctly with the ":" operator, even withoutThis is very useful for SNES, where preparing a DMA xfer requires one to know the bank number of the source/dest data.

(More information to be added soon)