vhelin/wla-dx #43: Make .RAMSECTIONs to use BANK in section placement

vhelin, 2015-08-02:
Currently .RAMSECTIONs occupy just slots, and don't use the BANK information for banking - WLA DX doesn't support banks for .RAMSECTIONs - this needs to be fixed as there are systems with RAM banks.

vhelin, 2015-08-10:
Any volunteers want to test my commit? :) I added a small test for this to examples/z80/ram_sections, but the more tests the merrier...

cr1901, 2015-08-11:
Will not be able to test until next week. I brought my local copy of the repo with me while I'm away handling family business, but... I just really haven't had time to look over things.

cr1901, 2015-08-15:
I just remembered that I was able to "fake" RAM support for LOROM SNES (only 8/128kB of RAM is shared between all banks), and that code that tries to reference RAM variables with the bank operator ":" will in fact reference the correct bank, long before this commit. However, this only worked for LOROM for some reason (despite .HIROM being a directive, HIROM doesn't have ANY RAM that is common to all banks).

I'm wondering if this commit fixes that? Will have to test soon.

vhelin, 2015-10-25:
Anyone tested the fix to this one?

Drenn1, 2016-02-21:
I know I'm really late here, but I finally removed the workaround I used in my project (having a separate slot for each RAM bank). Now all my ramsections are happily using the same slot with different banks, and the addresses remain correct.

closed 2019-09-06 by vhelin