vhelin/wla-dx #48: Cannot .define an alias for a RAMSECTION label/def

bazzinotti, 2015-08-26:
I've recently graduated to using RAM sections and I find them quite nice for automatically managing RAM across my project. However, I come from using .ENUM, and what I sometimes would do for temporary variables is temporarily define them with an alias.. Example: 
```
.enum $00
tmp1 db
.ende
; later in some routine
.define new_name tmp1
```
however
```
.ramsection "my-vars" bank 0 slot 8
tmp1 db
.ends
;later
.define new_name tmp1
```
does not work

I guess there is some aspect to a RAMSECTION that blocks what I'm trying to do. Did you intend on this behavior? Or is this considered a bug and should be fixed? If it is not considered a bug, what can I do?

Note: I am aware of the additional consideration of adding .undefine new_name after I'm finished with it.

nicklausw, 2015-08-26:
It's because  .enum provides a defined location for the label, but .ramsection doesn't, which confuses the assembler.

I'm pretty sure .sections don't have this problem though....might be worth looking into.

vhelin, 2015-08-26:
.ENUM creates internally .DEFINEs, but tmp1 in .RAMSECTION creates labels.
I just checked the code for .DEFINE, and it should accept stack objects, so
if you could try out

.define new_name tmp1+0

would it work? If yes, I'll need to check out why .define doesn't find tmp1
that was lexically present in the source code earlier... I might be
possible that the parser .DEFINE uses just ignores pure labels...

bazzinotti, 2015-08-26:
vhelin, you are correct, it works by adding `+ 0`

I would like to mention the README on .RAMSECTION is a little misleading because at its end it alludes that the .RAMSECTION produces .define statements. Perhaps consider adding a note in its documentation that it was only to illustrate the addresses that would be assigned to RAMSECTION variables, but that they are actually labels.

I am also having a new question on RAMSECTION labels -- if they are labels, how come their information is put into the symbol file ONLY if I add ':' after their declaration in the RAMSECTION?

bazzinotti, 2015-09-02:
I've experienced today a case where adding + 0 to the RAMSECTION label still caused problems. :[
`.define new_name tmp1+0` did not work and I don't know why.. Since I am in the context of a small-game engine it's hard to understand what caused it.

bazzinotti, 2015-09-04:
I've actually run into a case [in the context of my small PCE game engine -- so it's hard to pinpoint why] where using the + 0 trick did not work for me.. Could be a personal error.. Time will tell.

vhelin, 2015-09-04:
I'll try to give some time to WLA DX tomorrow or on Sunday. It's just that
the more I code the gig project that pays me some money, the more stuff
they want to add to the app. Means also bigger pay check, but less free
time... :P

bazzinotti, 2015-09-05:
Thanks Ville! I've prefixed the issues that bother me the most with exclamation points (!)

vhelin, 2015-09-05:
Weird, this works:

...

.MEMORYMAP
   DEFAULTSLOT     0
   ; ROM area
   SLOTSIZE        $4000
   SLOT            0       $0000
   SLOT            1       $4000
   SLOT            2       $8000
   ; RAM area
   SLOTSIZE        $2000
   SLOT            3       $C000
   SLOT            4       $E000
   SLOT            5       $FF01
.ENDME

.ROMBANKMAP
BANKSTOTAL 4
BANKSIZE $4000
BANKS 4
.ENDRO

.EMPTYFILL $AA

.RAMSECTION "RamSection 5 bank 1" BANK 1 SLOT 5 ALIGN 16
rs5b1_a db
rs5b1_b db
.ENDS

.define testor rs5b1_a

...

or at least I don't get a compile error. Cygwin / Windows 10, wla-z80...

vhelin, 2015-09-05:
With "does not work", what do you mean? :) A compile error, a linker error, a wrong value in output somewhere...? What error do you get?

vhelin, 2015-09-05:
If you pull the latest changes, there is now ram_sections test under examples/huc6280 - it compiles and works here on my computer, but does it work on yours?

bazzinotti, 2015-09-05:
I have limited time -- I did just pull and then try my own project again (not posted code) and it didn't work..
the error I get is same as before. linker error
```
FIX_REFERENCES: Reference to an unknown label "derp".
```

Your example code did compile and link though. *STRANGE*  I wonder what the difference is that is causing this issue?

bazzinotti, 2015-09-05:
...for
```
.define derp sx
    lda derp
```

```
.ramsection "myvars" bank 0 slot ZP_SLOT
frame  dw
sx dw
sy dw
vframe dw
joyparse_per_vframe db
.ends
```

sorry I cannot be more specific at this time.

bazzinotti, 2015-09-05:
I think the problem is that in your WLA huc6280 example, you need to try using the variable. .export is not enough (ie. try lda kalkaros_2) 

I will be mobile for the rest of the day good luck

nicklausw, 2015-09-05:
Be sure to define things in this order:

1. The ramsection
2. the alias
3. the alias reference

Because I did things in this order and got no errors referencing "kalkaros_2.w".

bazzinotti, 2015-09-05:
I always do in that order already

The ramsection 
the alias

bazzinotti, 2015-09-05:
I realize a mistake I made today. When recompiling my project after pulling in git.. I did not add + 0 as per the hack .. I'll try that when I get home..

vhelin, 2015-09-05:
Ok, I added

  lda kalkaros_2.w

to the end of the last section (pushed that to repo), and it compiles... ?

vhelin, 2015-09-05:
This is what I did, too, and got also no errors. :) Does this mean there is
no bug in WLA-DX? :)

Anyway, WLA-DX works like a train - it just parses stuff, converts that to
output data, and moves forward, lexically, in the source code. It cannot
handle (all) references into the future.

bazzinotti, 2015-09-05:
I will tag my engine's git repo as it is now so that we have a reference for future analysis of why this is happening..

bazzinotti, 2015-09-05:
I will make my engine available to you and public ASAP

vhelin, 2015-09-05:
Thanks, I guess it'll help! No need to make it public though...

cr1901, 2015-09-05:
I think we need an EBNF for WLA... but who's going to write that?

Well actually, that might not work. From what I can tell (Ville, correct me if I'm wrong), WLA uses a multiple-stack stack machine to do parsing. According to Wikipedia and [this SE answer](http://cs.stackexchange.com/questions/2832/is-a-push-down-automaton-with-two-stacks-equivalent-to-a-turing-machine), a stack machine with multiple stacks is equivalent to a Turing Machine.

Therefore, WLA's language may very possibly be Turing Complete, and it's not possible to actually for WLA to parse all valid programs.

Of course, the bugs that are being fleshed out may in fact be solveable by a Turing machine, and some state variables are just wrong. I think the only correct thing at this point is to handle bugs on a heuristic case-by-case basis.

bazzinotti, 2015-09-05:
I seriously consider wla dx inability to handle large object file (see the listing file segfault issue which arose as my object increased in size)... But I'm not sure

Also consider that I used to be able to use the +0 hack to define a reference a ramsection variable... Which, by the way, have you fixed?? Can I no longer require using + 0?  Thanks

nicklausw, 2015-09-05:
Uh...Bazz, this might be a dumb question, but are you REALLY sure you're using WLA's latest version? No one seems to be having this problem but you.

Sorry if this sounds insulting, I just don't get what is going on.

bazzinotti, 2015-09-05:
Yes

vhelin, 2015-09-05:
Anyway, Bazz, if you have the time and want to help me out here debugging
stuff, please make small projects under the (they're in the latest sources,
just pushed them) "bug_exhibition" directory. Then, after I fix a bug, I
move to the next project under that folder, and use it to hunt the next
bug... :) This way I don't have to spend time wondering how to make the
next bug surface, but can spend 100% of my time just fixing them bugs...

bazzinotti, 2015-09-06:
To rehash what's going on -- yes the huc6280 ram_sections example compiles/links fine for me too. This means that either there's an aspect of my project (which compiles and links fine without trying to .define an alias for RAMSECTION label) that creates a unique situation that causes a bug in WLA DX, or there's a bug in my code.. But the fact that my program appears to compile and run fine without aliasing a RAMSECTION label suggests bug in WLA DX. Keep in mind in the context of my project there are many RAMSECTIONS and once again, I am sourcing a lot of code into one object file...

any bugs that I find that are easy to segment into a small bug_exhibition example I will gladly wrap up for you.. Where do I submit the examples tho? (and how)

bazzinotti, 2015-09-06:
I guess I can zip them up onto an HTTP server for you guys..

bazzinotti, 2015-09-06:
Currently documenting my engine as best I can so u guys can check it out

bazzinotti, 2015-09-06:
http://botb.club/~bazz/wladx-pce-engine.zip

There, see README.txt -- compile 'game' -- then you can try adding a .define yourself and watch it fail.

nicklausw, 2015-09-06:
I think the solution to your problem is to .export all your aliases, that fixed the problem for me.

bazzinotti, 2015-09-06:
Nicklausw you are right!! But why?? Why is this needed? As I understand .EXPORT makes the symbol visible to all object files, but since I only have one...

bazzinotti, 2015-09-06:
Furthermore, if exporting is necessary (which I consider a bug) -- how can I alias RAMSECTION variables that are *not* supposed to be exported (made accesible globally)

bazzinotti, 2015-09-06:
I think there's a bug in WLA DX on that level.

bazzinotti, 2015-09-06:
It is not right to have to force aliases to have global scope across objects [to restate the issue]

bazzinotti, 2015-09-07:
This is still an issue. I will create a bug_exhibition for it tomorrow.

bazzinotti, 2015-09-07:
To be discrete, this issue is now turning into -- when a single-object has a ramsection followed by one of its labels being aliased by a .DEFINE statement, that define must be exported into order to be visible AT ALL. This goes against the normal behavior -- where the definition should be usable from inside that object, and only should have to be exported for visibility in other objects. 

However, Ville did attempt to eliminate the need for adding `+ 0` to the definition. So that's good.. Someone please advise what to do next (I will still create a bug exhibition tomorrow)

bazzinotti, 2015-09-07:
http://botb.club/~bazz/wladx/issue48.zip -- there's a premise in main.s explaining

vhelin, 2021-04-29:
I tested this again using the latest sources and indeed 

.ramsection "my-vars" bank 0 slot 8
tmp1 db
.ends
;later
.define new_name tmp1

assembles just fine. Also

.define derp sx
    lda derp

.ramsection "myvars" bank 0 slot ZP_SLOT
frame  dw
sx dw
sy dw
vframe dw
joyparse_per_vframe db
.ends

works without any problems.

http://botb.club/~bazz/wladx/issue48.zip doesn't exist any more, so I guess this can be closed.

closed 2021-04-29 by vhelin