vhelin/wla-dx #49: .w doesn't work for $0000 unless adding + 0

bazzinotti, 2015-08-26:
example: 
```
.enum $0000
derp DW
.ende
; later in some routine
lda derp.w ; this will load as from zero-page! bad!
lda derp.w + 0 ; this will load as absolute. Good boy!
lda derp.b + 0 ; this will load as zero page. Good boy!
lda derp.b       ; this will load as zero page. Good
```

I've tried using `enum $0000.w` which does not help

vhelin, 2015-08-26:
Which WLA DX assembler are you using? Sounds bad...

bazzinotti, 2015-08-26:
I checked out the git repo about last week or two weeks ago-ish

vhelin, 2015-08-26:
I mean, is it 6502, Z80... ?

bazzinotti, 2015-08-26:
huc6280

vhelin, 2015-08-26:
This really sounds like a bug in the parser, because if you give .w if
should be .w all the way until the very end... I'll need to check this one
out, but later today, now I'm trying to get some work done. :)

bazzinotti, 2015-09-06:
I won't make a bug_exhibition for this unless requested.

vhelin, 2015-09-06:
That's ok, I can test this without one. I'll start to work on this now...

bazzinotti, 2015-09-06:
Ville, since your time is limited and appreciated, can you focus on https://github.com/vhelin/wla-dx/issues/53 -- it really bothers me

vhelin, 2015-09-06:
Sure...

bazzinotti, 2015-09-06:
vhelin -- Thank you - followed by https://github.com/vhelin/wla-dx/issues/52  -- these are the worst from a development standpoint.

bazzinotti, 2015-09-07:
I made a bug_exhibition for this anyways, incase others wanted to work with it : http://botb.club/~bazz/wladx/issue49.zip

fseidel, 2018-03-17:
The bug seems to go away for me when a label is used instead of a zero constant.

vhelin, 2021-03-15:
I tried the example code, but today it assembles like it should so I assume that this was fixed at some point, but no-one came to this thread to report about it... If you still see this bug, please reopen the issue and I'll try to act a little faster. :)

closed 2021-03-15 by vhelin