vhelin/wla-dx #50: Macro - check if an argument is an immediate value

bazzinotti, 2015-09-01:
Is it possible to do this in WLA DX? [see title]
ie. #$4000 vs. $4000

vhelin, 2015-09-01:
I don't think so. The assembler itself should find that # and work
accordingly, but if the user would want to do the test himself - I don't
think so. How would you like to do the testing?

bazzinotti, 2015-09-01:
I am porting some code which handles the macro like so: 
```
.macro vreg
	lda	\1
	sta	<vdc_reg
	.if (\?1 = ARG_IMMED)
	st0   \1
	.else
	sta	video_reg
	.endif
.endm
```

Here's another example: 
```
.macro spr_pattern
    ldy   #4
    .if (\?1 = ARG_IMMED)
     lda   #<((\1) >> 5)
     sta   [_si],Y
     lda   #>((\1) >> 5)
    .else
     lda   \1
     sta   [_si],Y
     lda   \1+1
    .endif
    iny
    sta   [_si],Y
.endm
```

cr1901, 2015-09-02:
Bazz wants this equivalent NASM feature in WLA: http://www.nasm.us/doc/nasmdoc4.html#section-4.4.6

dwsJason, 2020-02-27:
+1 on this feature

vhelin, 2020-02-28:
So you want this to work inside a macro?

  .if (\?1 = ARG_IMMED)

Are the following types enough? WLA already knows these types at the moment:

- integer
- floating point number
- string
- address label
- pending calculation

... or should we just set the type to "number" when getting an integer or a floating point number?

dwsJason, 2020-02-28:
Here's what I want this for:

.macro PushLong ARGS value
   .if(value = ARG_IMMED)
      pea value>>16
      pea value
   .else
     lda value+2
     pha
     lda value
     pha
  .endif
.endm

I want to be able to tell the difference between a memory address, and an immediate constant.   I could see some value in in detecting strings (since you could maybe define them into a section, then reference their address).   Or if there was functionality that would let you .rept through the string character by character in order to do manipulation, like cat, or to recode the string into 5 bits per character).     I feel like string features are a whole new type of feature .    The ability to determine if the value is immediate, or an address would be nice.

vhelin, 2020-02-28:
Hey, please open a new issue for that ".REPT through a string" enhancement. Sounds useful, but a totally separate issue.

Kroc, 2020-02-28:
Also, be aware of index registers; how would a macro handle an optional ", x" after the first parameter?

vhelin, 2020-02-28:
You can also use ``\?`` to ask for the type of the argument in the
following fashion::

    .macro .differentThings
      .if \?1 == ARG_NUMBER
        .db 1
      .endif
       .if \?1 == ARG_STRING
        .db 2
      .endif
      .if \?1 == ARG_LABEL
        .db 3
      .endif
      .if \?1 == ARG_PENDING_CALCULATION
        .db 4
      .endif
    .endm

    .section "TestingDifferentThings"
    TDT1:
        .differentThings 100
        .differentThings "HELLO"
        .differentThings TDT1
        .differentThings TDT1+1
    .ends

The previous example will result in .db 1, 2, 3, 4

vhelin, 2020-02-28:
How is it working for you guys?

vhelin, 2020-02-28:
> Also, be aware of index registers; how would a macro handle an optional ", x" after the first parameter?

We'll just need to test it. I hope it works after my latest commit. :) This is engineering.

Kroc, 2020-02-29:
How do we check if an argument exists or not?

vhelin, 2020-03-01:
Use NARGS?

"Every time a macro is called a definition NARGS is created. It shows only
inside the macro and holds the number of arguments the macro was called
with."

vhelin, 2021-04-27:
It's funny that I totally missed the original point and did something else here. :D Anyway, with the latest sources this issue should now be fixed...

closed 2021-04-28 by vhelin