vhelin/wla-dx #51: WLA DX IRC/Forum community?

bazzinotti, 2015-09-01:
Is there an IRC channel or WLA Forum community?

vhelin, 2015-09-01:
I don't know of any... :(

bazzinotti, 2015-09-01:
I will start idling in FreeNode / ESPER #wladx

nicklausw, 2015-09-01:
http://network54.com/Hide/Forum/56397 there was one. Probably can't still be used.

cr1901, 2015-09-02:
I'm not going to be able to work on WLA for a while. So if you can do pull requests (and make sure they're isolated to one functional change only!), or Ville can fix bugs, IRC might be a good way for more ppl to understand WLA internals.

It's getting there, but still sorely needs a refactoring.

bazzinotti, 2015-09-02:
I am not scared of doing pull requests. I have experience separating mods into small logical changes on the code base. 

Based on not having touched the WLA DX code at all && my recent-related issues -- does anyone have any pointers on relevant files / paradigms that would help me out? Thanks

bazzinotti, 2015-09-02:
To be honest I'm a bit overwhelmed by adding "modifying WLA DX" to my already-full plate. I think for the sake of decreased stress levels I will solely be an issue submitter for now.

vhelin, 2015-09-02:
I'm busy doing freelance coding at the moment, which is nice, I've been
unemployed for quite some time, and I might even get a real job later this
month! :) Anyway after I get rid of this flu I'm currently enjoying I'll
start checking out the bug reports / feature requests. Keep on submitting
the issues, it helps! :)

Thanks!

nicklausw, 2015-09-02:
I'd help contribute but am not very good at C (why I code in ASM with WLA, of course) and the code base doesn't make a whole lot of sense to me.

If there's one thing I've noticed, it's that pass_1.c is redundant in some places. Could certainly help shrink WLA's size to have some extra functions for things done a lot.

bazzinotti, 2015-09-03:
I wanted to state here [I'm treating this thread as a kind of general thread now] that all of the bugs I've been posting -- although personally currently experienced while programming for huc6280 -- I can recall a lot of the same/similar issues in the past on the 65816. But I developed on the 65816 mostly with the WLA DX homepage release archives. However my gut tells me the problems experienced are universal across the CPUs.. I am hoping Ville/other WLA DX programmer expert, can tell me whether this is case as he solves the issues! [tell me whether the fixes are universal or only to a certain CPU]. Thx

nicklausw, 2015-09-03:
Last time I used WLA-6502, the .w fix was always needed but none of that "+ 0" stuff.

Ever feel like WLA just needs a total do-over? I get that feeling sometimes.

vhelin, 2015-09-03:
> Last time I used WLA-6502, the .w fix was always needed but none of that
> "+ 0" stuff.
>
> Ever feel like WLA just needs a total do-over? I get that feeling
> sometimes.

Nope. Feel free to do it. :) I don't have the time or energy anymore to
give away for free... Can only maintain WLA DX sometimes.

nicklausw, 2015-09-03:
XD trust me, I wasn't implying that any of us would have time to do it.

cr1901, 2015-09-03:
Rewrites in general are a horrible idea anyway. Just look at what happened to Netscape. Like I said, refactoring is needed; it's also probable that these bugs existed for years, but either nobody flushed them out before Ville decided to start working on it again (and I broke the SVN repo/we migrated to Github :P).

There are some SourceForge bugs I should probably open issues for/migrate here.

nicklausw, 2015-09-04:
How'd you break the SVN repo?

Personally I'm happy that WLA DX was moved to GitHub. Can't quite figure out Sourceforge, too "clunky".

bazzinotti, 2015-09-06:
I'm starting to confuse the difference between a label's scope between sections and between object files.. 

Please correct the following understanding I have: 

As I expect it, all labels within all sections of an object file, except those starting with '_' are visible to any section of that object file, but not to any other object file and its sections. Is that right?

bazzinotti, 2015-09-07:
Guys, at the end of the day, ville made progress, I talked a lot but I actually segmented every issue into its own little sandbox (I'm not full of s*** -- I escaped my project code). All of the issues bug_exhibitions can be found at http://botb.club/~bazz/wladx

lhsazevedo, 2021-01-13:
Well, [Github Discussions](https://docs.github.com/en/free-pro-team@latest/discussions) are now in [public beta](https://github.blog/changelog/2020-12-08-github-discussions-now-available-as-a-public-beta/).

vhelin, 2021-03-15:
Thanks for the tip, @lhsazevedo, and now we have a discussion forum here @ Github: https://github.com/vhelin/wla-dx/discussions

closed 2021-03-15 by vhelin