vhelin/wla-dx #52: Hashtag(imm. value) not being parsed correctly in call to macro

bazzinotti, 2015-09-01:
From my last issue filed today (associated with macros), https://github.com/vhelin/wla-dx/issues/50
Ville wrote: `The assembler itself should find that # and work
accordingly`

This is not the case (huc6280) with such an example: 
```
.macro sub          ; subtract byte-sized value
     sec
     sbc    \1
.endm

; later in a code section
sub   #$20
```

The error is `STACK_CALCULATE: Syntax error. Invalid use of modulo`

The reason this bothers me is because it's blocking WLA DX from being able to use one macro to potentially handle different kind of arguments all in one macro (convenience) (RAM/ROM access instead of immediate value). 
I am hoping to *not* be recommended a workaround (2 macros). I'd like to see or help WLA DX become awesome -- instead of working around its problems.

bazzinotti, 2015-09-04:
This is probably the most aggravating bug of them all.. I'm trying to simplify my work with macros but it's irritating to have to create separate macros based on immediate/address knowing full well that if WLA DX was fixed I would never have had to create these separate macros and separate calls to them. 

But I don't know where to start in fixing this issue :[ if anyone doesn't have the time but has the know-how -- a pointer or three or four would be helpful. Thank you

bazzinotti, 2015-09-04:
This is probably the most aggravating bug of them all.. I'm trying to simplify my work by using macros but it's irritating to have to create separate macros based on immediate value/address access knowing full well that if WLA DX was fixed I would never have had to create these separate macros and separate calls to them. (the .w bug https://github.com/vhelin/wla-dx/issues/53 is also very related -- as it means having to force create different kinds of macros based on their ZP/ABS address access, when this could be automated!)

But I don't know where to start in fixing this issue :[ if anyone doesn't have the time to fix the whole issue but has the know-how -- a pointer or three or four would be helpful. Thank you. Otherwise I will try to wait.

I really want to fix these issues so I can code my engine the right way!! I'm X_X otherwise.

bazzinotti, 2015-09-06:
bug_exhibition : http://botb.club/~bazz/wladx/issue52.zip

vhelin, 2015-09-06:
Oh, now I can see why the assembler doesn't find that #... :) It's part of a string argument to a macro! I bet

...

.macro sub          ; subtract byte-sized value
     sec
     sbc    #\1
.endm

; later in a code section
sub   $20

...

would work. Now in the problematic case the macro is given a string, not a value, Or actually the parser thinks that it's a calculation that begins with a modulo, not a string.

Hmm. I guess I should first fix the parser to interpret "#$20" as a string, and then make the opcode decoder understand macro string arguments, unless it already thinks they are all labels. Tricky. I need to think about this, fixing it is not obvious to me at the moment...

bazzinotti, 2015-09-06:
The whole point from my perspective is that I want 1 macro that can parse immediate value arguments and address arguments.

ie. I want to be able to do both: 
```
sub $20       ; subtract zero-page value at $20
sub $2000   ; subtract value at $2000
sub #$20     ; subtract immediate value #$20
```

bazzinotti, 2015-09-06:
This gets a bit hard to do when you consider RAMSECTION variables as an argument.. which for some reason can only be accessed as a WORD when .w is specified :\ -- making it impossible to write a generic macros as of right now..

So it's that problem ^ and also this issue #52 that are making generic macros impossible for me.

bazzinotti, 2015-09-06:
Let me know if I should file a new issue I can include a new bug_exhibition test and call the issue "generic macro" .. I will put in a decent test that probably won't compile for awhile :P

bazzinotti, 2015-09-06:
Of course, that's contrary to my hopes. I hope we can fix it ASAP

bazzinotti, 2015-09-06:
Perhaps the "generic macro test" could be a combination of the specs I'm looking to satisfy followed by whatever other specs you / others expect a macro to handle :)

vhelin, 2015-09-06:
I could fix this issue "relatively" easy by introducing this as a special case in the opcode parser, but that would be just plain nasty in programming / architecture sense. :) Anyway, this one needs some time. I might be able to fix it later today, or not. Let's see...

bazzinotti, 2015-09-06:
:smiley:

bazzinotti, 2015-09-06:
bug_exhibition test for use when issue 52/53 are solved : http://botb.club/~bazz/wladx/generic_macro_test.zip

mipani, 2016-01-05:
Seconding the ability to pass the immediate operator as part of the argument! I have the same issue, putting together my own SNES framework. Would be very nice!

nicklausw, 2016-01-07:
I noticed that the stack calculator said not to accept the modulo if it was the first symbol read, which is why #$80 won't work. Added a fix that should make it so the modulo is only accepted as the first symbol.

vhelin, 2016-01-07:
Sorry guys, a few days ago I was told about this wonderful surprise
deadline that we'll (or mostly I) have next Monday, 11th of January 2016,
so I can't help you guys much before that, unless all the remaining issues
we have in this software project get solved with the first try. :)

Anyway, next week latest I should really have time to give to WLA DX
again... But if e.g. mr. nicklausw can help you out - all the better. :)

nicklausw, 2016-01-07:
With that said, would you guys test out the fix_65 branch of my wla-dx fork? Be sure to go to the fork and not wla65, as that's my wla "derivative" that I've begun work on. (Without permission to change the name. I'm not even sure if the GPL 2.0 permits name change. Prolly shoulda thought that through a bit more...)

vhelin, 2021-04-28:
Commit https://github.com/vhelin/wla-dx/commit/534342952f5154cdbec548819a87abe5dce980c7 made it so that now you can prefix whatever you feed a .MACRO as an argument with a '#', the system now eats that away and you can now ask .if (?1 = ARG_IMMEDIATE) inside the .MACRO.

That fixed GitHub issue #50 and should also fix this issue...

closed 2021-04-28 by vhelin