vhelin/wla-dx #54: in macro: adding .w to parameter requires adding '+ 0'

bazzinotti, 2015-09-02:
example:

```
.macro derp
lda \1.w
.endm

; in code
derp $100
```
`ERROR: Unknown symbol "lda".`

temporary fix: 

```
lda \1.w + 0
```

vhelin, 2015-09-05:
This doesn't compile (06-Sep-2015), and I need to dig deep into the WLA's parser. :) Can't fix this today, but I'm on it!

Dang, the code I wrote 10+ years ago, I wish I had been smarter back then... :)

bazzinotti, 2015-09-05:
:)

cr1901, 2015-09-05:
CMake has a built-in testing framework called CTest. However, I like Ville's idea better because it's essentially platform-independent and can also be integrated into CTest as well as a generic Makefile that chokes in WLA fails to compile in any way, shape, or form.

Kinda like LLVM, perhaps all bugs from now on should become part of regression testing?

vhelin, 2015-09-05:
> CMake has a built-in testing framework called CTest. However, I like
> Ville's idea better because it's essentially platform-independent and can
> also be integrated into CTest as well as a generic [...]

Sounds good - after the bug in WLA-DX is fixed and the small project
compiles and links under the bug_exhibition directory, the whole project
can be moved to examples (should be renamed to tests?) dir and added to
run_unit_tests.sh script.

The tests aren't really unit tests, so that script might also need a name
change, but anyway, food for thought. :)

cr1901, 2015-09-05:
I think there should be a tests and examples directory. Where the examples are minimum component examples that test WLA features that run on real systems :P.

vhelin, 2015-09-06:
cr1901: You are right, tests should be tests (can contain whatever ascii junk, and look terrible) and examples should be examples (clean and nice, even useful examples, perhaps template projects).

About this bug:

.macro derp
lda \1
.endm

this works, but

.macro derp
lda \1.w
.endm

does not as the parser doesn't understand "\1.w" - it correctly decodes the "\1" part, but it doesn't understand that the ".w" is part of "\1". But

.macro derp
lda \1
.endm

; in code
derp $100.w

... this should work. :)

bazzinotti, 2015-09-06:
The reason is another bug: https://github.com/vhelin/wla-dx/issues/53 .. Fix 53 and I can probably adapt to the way WLA DX likes to see it. But in this case we should document the README to warn the user about the macro's pickiness. (ie. derp $100.w vs. \1.w)

bazzinotti, 2015-09-06:
You might be thinking what I'm thinking -- "If bazz called the macro with .w in the argument, shouldn't the macro work even tho it only says \1 without .w?"  -- This is sound thinking.. I'll make a bug exhibition on it

vhelin, 2015-09-06:
Nah, as this seems to confuse people it'd be better to make input_number() in parse.c to eat and use the .b/.w that follows a macro argument. :) I'll do this later today.

bazzinotti, 2015-09-06:
What I'm about to document in this post can be considered a bug likely based on issue #53 -- I put it here because it explains why I was forced to used `\1.w` in my macro(s). Throw this in huc6280 bug exhibition main.s and you'll see: 
```
.ramsection "vars" bank 0 slot 1
derp dw
.ends

.macro sub          ; subtract byte-sized value
	sec
	sbc \1
.endm

.bank 0 slot 0
.org 0
.section "code"
	sub derp.w
.ends
```

```
main.o:main.s:32: FIX_REFERENCES: Value ($2000) of "derp" is too much to be a 8bit value.
```

vhelin, 2015-09-06:
Well, pull the latest sources, and

  lda \1.w

and

  sbc \1.w

will work. :)

bazzinotti, 2015-09-06:
confirmed here - it works.

I recall thinking to myself -- I hope Ville is not just hacking this
feature in and creating more bugs or in a badly designed fashion.

closed 2015-09-06 by vhelin