vhelin/wla-dx #55: wlalink -vSi occasionally segfaulting

bazzinotti, 2015-09-02:
OS X Yosemite 10.10.4 using WLA DX trunk --

I've been compiling for a week with no problems until now. 

**wlapce**
```
#!/bin/sh
if [ "$1" = "" ]; then
	echo "Usage example: `basename $0` derp.s"
	printf "\toutputs: derp.pce, derp.sym, derp.lst\n"
	exit 1
fi

fn_noext="`basename ${1%.*}`"
echo '[objects]' > link
echo $fn_noext.o >> link

wla-huc6280 -vxio $1 $fn_noext.o
wlalink -vSi link $fn_noext.pce

rm $fn_noext.o
```

I think this is due to a bad, yet valid approach to WLA DX, in which I am creating one large single object file and .include'ing my entire project into it. --> This means lots of listing output files and it was working until today.. now the linking process *occasionally* experiences a couple-second-delay-followed-by-segfault if I continue to produce listing output.. This bug is low priority in my eyes.

I'm not sure how to reproduce the bug unfortunately and lack the will to continue debugging this. 

```
Program received signal SIGSEGV, Segmentation fault.
0x00000001000116eb in listfile_write_listfiles (e=0x100131640)
    at /Users/bazz/Code/wla-dx/wlalink/listfile.c:159
159		  if (b[m] == 0xA) {
(gdb) p/x m
$1 = 0x7ff000
(gdb) x/s b
0x101001000:	";.include \"video-core.i\"\n.ifndef ASCII_S\n.define ASCII_S\n.ramsection \"ascii-ram\" bank 0 slot 1\nfont_loc: dw\nfont_wpt: dw\nfont_bpt: dw\n.ends\n.bank 0 slot 7\n.section \"ascii\"\n; uploads font to tile_vram_"...
(gdb) x/s b[m]
Cannot access memory at address 0x101800000
(gdb)
```

I guess WLA DX has a single-buffer to process the file and it's not enough. [I feel I may be wrong]

bazzinotti, 2015-09-06:
Oddly enough, in the current state of my engine (as in the zip file http://botb.club/~bazz/wladx-pce-engine.zip), this problem has disappeared.. Is it worth digging into the commit history to find a state where the bug is active?

bazzinotti, 2015-09-07:
This is another bug I cannot reproduce even with an old version of WLA DX (before ville started committing this week).. Hmph *laughs* Since I don't know what happened, maybe we can close this too. Up to you.

cr1901, 2015-09-07:
The -vSi command line parameters break one of the test suite examples as I recall, so it's a legit bug.

bazzinotti, 2015-09-07:
actually - I did a `make clean` followed by a `make` on WLA DX after reverting to older commit and yes the bug is back. And yes, I re-checked all my observations on Bug #59 with a `make clean` before making the latest commit -- but those observations remain the same

vhelin, 2015-09-07:
Oh man, I've totally forgotten to keep an eye on the older tests in the examples directories. I'll definitely check them out later this week. I see there two failing tests, and two tests with makefile problems. Dang...

vhelin, 2016-02-02:
Guys, anyone, if you can supply me with a project that crashes WLA DX, please do so - currently I cannot repeat these crashes with my own projects thus debugging is pretty much impossible, given the physical resource limits (time, energy, ...) I have here at my disposal. :)

I can add plenty of checks to wlalink/listfile.c:listfile_write_listfiles(), later when I've taken care of bigger issues, so line 159 crash can perhaps be avoided in the code without any hard signals to the exe, but I'd love to nail this bug asap. I just hate crash bugs. :)

It sounds and looks like wlalink crashes, because the index m goes out of bounds, but I'd love to know the reason for that.

vhelin, 2021-03-17:
No new reports about this issue since 2016 so I think we can close the issue. If you still can get WLA/WLALINK to crash, please submit a new issue.

(gdb) p/x m
$1 = 0x7ff000

that m is pretty big, but it's not _too_ big. It could be that the listfile creator function has somehow ran past the end of the source file and now it's just poking random memory, but as I cannot dig deeper into this one using the supplied information we just need to wait for the bug to reappear.

closed 2021-03-17 by vhelin