vhelin/wla-dx #58: Macro ARGS Issue #2

bazzinotti, 2015-09-04:
after fixing the macro posted in https://github.com/vhelin/wla-dx/issues/57 another ARGS error: 
```
FIX_REFERENCES: Bank number request for an unknown label "label".
```
.macro UP_VRAM_ADDR ARGS label, vram_addr, bytes
    lda #:label ;LABEL
    sta <bl
    lda #<label
    sta <si
    lda #>label
    sta <si + 1
    lda <vram_addr
    sta <di
    lda >vram_addr
    sta <di + 1
    lda #<(\3/2) ; words
    sta <cl
    lda #>(\3/2) ; words
    sta <ch
    jsr load_vram
.endm
```

temporary fix:
```
.macro UP_VRAM_ADDR ARGS label, vram_addr, bytes
    lda #:\1 ;LABEL
    sta <bl
    lda #<label
    sta <si
    lda #>label
    sta <si + 1
    lda <vram_addr
    sta <di
    lda >vram_addr
    sta <di + 1
    lda #<(\3/2) ; words
    sta <cl
    lda #>(\3/2) ; words
    sta <ch
    jsr load_vram
.endm
```

this makes me scared that label and \1 are different, so I'll probably just avoid using ARGS from now on until fixes / someone sheds light on the issue

bazzinotti, 2015-09-04:
I did a lot of editing to the post, so if you're reading email -- please see the github website URL for latest version.

bazzinotti, 2015-09-04:
corrected the code

bazzinotti, 2015-09-07:
bug_exhibition :) http://botb.club/~bazz/wladx/issue58.zip

bazzinotti, 2015-09-07:
Hey, issue #58 (this issue) and #59 both stem from an inability to access the bank byte. Is there a relationship?

vhelin, 2015-09-07:
If you give names to macro arguments, they are going to be... tadah - internally definitions! :) \1 is parsed differently...

Why I didn't create definitions called \1 etc - I do not kwow. That would be simpler... Well, I'll add this to my do-when-refactoring list. :) Live and learn.

Anyway, I'll look into this issue.

vhelin, 2015-09-08:
In the end I went and rewrote the macro argument expansion function, and it should be pretty sweet now. :) I even had labels like "\1_\2:" in my old examples projects that were failing... I had no memory of supporting such usage of macro arguments, especially having macros calling macros, but all that should work now.

If you could test this, thanks! :)

bazzinotti, 2015-09-08:
I can't do this right now, but I'll do it after I take care of some very important things.

bazzinotti, 2015-09-09:
I will be going offline for the next few days up to a week. I did not get the chance to test this.

nicklausw, 2015-09-09:
I can test it later today.

vhelin, 2015-09-09:
Cool, no hurry guys! I can also try to test it more later today, if I have
time... I'll start working in a new job next Monday, so after that my time
will be pretty limited, at least for a month or two until I get everything
rolling...

vhelin, 2015-10-25:
Anyone tested the fix to this?

bazzinotti, 2015-11-05:
Hi Ville, I'm getting a new job so I'm pretty busy. But I've been thinking to myself -- how can we reliably test without a testing framework?? There's no real way to see if previous functionality is being broken in any way, right? If that's the case, we really ought to bang out test cases before being able to implement new features reliably. Am I right?

cr1901, 2015-11-06:
cmake provides ctest. I had meant to port all the examples to use test (or be usable from Makefiles alone), but for the time being my contributions to WLA are limited thanks to real life problems. I'll pull the code when I can and add prelim ctest support.

nicklausw, 2015-11-10:
...Aw crap, I meant to test this two months ago. Will do if I can remember (on a school computer at the moment, so wla's not handy, obviously).

vhelin, 2016-02-02:
I tested this one again today, using the latest sources, and seemed to work, so I'll close this issue 58.

closed 2016-02-02 by vhelin