vhelin/wla-dx #59: Nested macro calls sharing parameter caused wla-huc6280 to segfault

bazzinotti, 2015-09-05:
I used to have a macro as follows: 
```
; ADDR suffix because WLA DX cannot take in a argument that could be # or $
.macro UP_VRAM_DBANK_ADDR ARGS label, vram_addr, bytes
    ; calculate PCE bank from WLA databank
    lda #>(\1 >> 5)    ; converts page-address into bank number
    clc
    adc #:\1    ; proper bank number
    sta <bl

    lda #<\1
    sta <si
    lda #>\1
    and #$1f    ; mask out page
    sta <si + 1

    lda \2.w + 0
    sta <di
    lda \2.w + 1
    sta <di + 1
    lda #<(\3>>1) ; bytes
    sta <cl
    lda #>(\3>>1) ; bytes
    sta <ch
    jsr load_vram
.endm
```

It would work fine.. I then ended up refactoring it into: 
```
; IN : ARG1 == LABEL from Databank
; OUT: A == PCE bank number
.macro CALC_PCE_BANK_FROM_WLA_DATABANK
    lda #>(\1 >> 5)    ; converts page-address into bank number offset
    clc
    adc #:\1    ; get proper bank number
.endm

; ADDR suffix because WLA DX cannot take in a argument that could be # or $
.macro UP_VRAM_DBANK_ADDR ;ARGS label, vram_addr, bytes
    CALC_PCE_BANK_FROM_WLA_DATABANK \1
    sta <bl

    lda #<\1
    sta <si
    lda #>\1
    and #$1f    ; mask out page
    sta <si + 1

    lda \2.w + 0
    sta <di
    lda \2.w + 1
    sta <di + 1
    lda #<(\3>>1) ; bytes
    sta <cl
    lda #>(\3>>1) ; bytes
    sta <ch
    jsr load_vram
.endm
```

But this causes a segfault: 
```
Program received signal SIGSEGV, Segmentation fault.
_expand_macro_arguments (in=0x7fff5f4001d0 ":\\1", expands=0x7fff5f4001ac)
    at /Users/bazz/Code/wla-dx/parse.c:769
769	  for (d = 0; in[i] != 0; i++) {
```
```
(gdb) bt
#0  _expand_macro_arguments (in=0x7fff5f4001d0 ":\\1", expands=0x7fff5f4001ac)
    at /Users/bazz/Code/wla-dx/parse.c:769
#1  0x0000000100006293 in _expand_macro_arguments (in=0x7fff5f400380 ":\\1",
    expands=0x7fff5f40035c) at /Users/bazz/Code/wla-dx/parse.c:801
#2  0x0000000100006293 in _expand_macro_arguments (in=0x7fff5f400530 ":\\1",
    expands=0x7fff5f40050c) at /Users/bazz/Code/wla-dx/parse.c:801
#3  0x0000000100006293 in _expand_macro_arguments (in=0x7fff5f4006e0 ":\\1",
    expands=0x7fff5f4006bc) at /Users/bazz/Code/wla-dx/parse.c:801
#4  0x0000000100006293 in _expand_macro_arguments (in=0x7fff5f400890 ":\\1",
    expands=0x7fff5f40086c) at /Users/bazz/Code/wla-dx/parse.c:801
#5  0x0000000100006293 in _expand_macro_arguments (in=0x7fff5f400a40 ":\\1",
    expands=0x7fff5f400a1c) at /Users/bazz/Code/wla-dx/parse.c:801
#6  0x0000000100006293 in _expand_macro_arguments (in=0x7fff5f400bf0 ":\\1",
    expands=0x7fff5f400bcc) at /Users/bazz/Code/wla-dx/parse.c:801
#7  0x0000000100006293 in _expand_macro_arguments (in=0x7fff5f400da0 ":\\1",
    expands=0x7fff5f400d7c) at /Users/bazz/Code/wla-dx/parse.c:801
#8  0x0000000100006293 in _expand_macro_arguments (in=0x7fff5f400f50 ":\\1",
    expands=0x7fff5f400f2c) at /Users/bazz/Code/wla-dx/parse.c:801
#9  0x0000000100006293 in _expand_macro_arguments (in=0x7fff5f401100 ":\\1",
    expands=0x7fff5f4010dc) at /Users/bazz/Code/wla-dx/parse.c:801
#10 0x0000000100006293 in _expand_macro_arguments (in=0x7fff5f4012b0 ":\\1",
    expands=0x7fff5f40128c) at /Users/bazz/Code/wla-dx/parse.c:801
#11 0x0000000100006293 in _expand_macro_arguments (in=0x7fff5f401460 ":\\1",
    expands=0x7fff5f40143c) at /Users/bazz/Code/wla-dx/parse.c:801
---Type <return> to continue, or q <return> to quit---
#12 0x0000000100006293 in _expand_macro_arguments (in=0x7fff5f401610 ":\\1",
    expands=0x7fff5f4015ec) at /Users/bazz/Code/wla-dx/parse.c:801
#13 0x0000000100006293 in _expand_macro_arguments (in=0x7fff5f4017c0 ":\\1",
    expands=0x7fff5f40179c) at /Users/bazz/Code/wla-dx/parse.c:801
#14 0x0000000100006293 in _expand_macro_arguments (in=0x7fff5f401970 ":\\1",
    expands=0x7fff5f40194c) at /Users/bazz/Code/wla-dx/parse.c:801
#15 0x0000000100006293 in _expand_macro_arguments (in=0x7fff5f401b20 ":\\1",
    expands=0x7fff5f401afc) at /Users/bazz/Code/wla-dx/parse.c:801
#16 0x0000000100006293 in _expand_macro_arguments (in=0x7fff5f401cd0 ":\\1",
    expands=0x7fff5f401cac) at /Users/bazz/Code/wla-dx/parse.c:801
#17 0x0000000100006293 in _expand_macro_arguments (in=0x7fff5f401e80 ":\\1",
    expands=0x7fff5f401e5c) at /Users/bazz/Code/wla-dx/parse.c:801
#18 0x0000000100006293 in _expand_macro_arguments (in=0x7fff5f402030 ":\\1",
    expands=0x7fff5f40200c) at /Users/bazz/Code/wla-dx/parse.c:801
#19 0x0000000100006293 in _expand_macro_arguments (in=0x7fff5f4021e0 ":\\1",
    expands=0x7fff5f4021bc) at /Users/bazz/Code/wla-dx/parse.c:801
#20 0x0000000100006293 in _expand_macro_arguments (in=0x7fff5f402390 ":\\1",
    expands=0x7fff5f40236c) at /Users/bazz/Code/wla-dx/parse.c:801
#21 0x0000000100006293 in _expand_macro_arguments (in=0x7fff5f402540 ":\\1",
    expands=0x7fff5f40251c) at /Users/bazz/Code/wla-dx/parse.c:801
#22 0x0000000100006293 in _expand_macro_arguments (in=0x7fff5f4026f0 ":\\1",
    expands=0x7fff5f4026cc) at /Users/bazz/Code/wla-dx/parse.c:801
```

there's over 4000 of these recursive function calls.

bazzinotti, 2015-09-05:
the problematic line is the one that tries to get the bank byte

bazzinotti, 2015-09-05:
temporary work-around: 

```
.macro CALC_PCE_BANK_FROM_WLA_DATABANK
    lda #>(\1 >> 5)    ; converts page-address into bank number offset
    clc
    adc #\2    ; get proper bank number
.endm
```

```
CALC_PCE_BANK_FROM_WLA_DATABANK \1 :\1
```

vhelin, 2015-09-06:
Right, the parser would just get stuck where you gave \1 as a parameter to a macro that used \1... Recursive parser didn't change the macro context so it would just repeatedly replace that lowlevel \1 with a toplevel \1... :)

Anyway, if you pull the latest sources it should work now. This was a real bug in the parser. My brain hurts now, need some sleep. :) And might drink a beer or two before that...

Anyway, if it doesn't work, let me know asap. I still have the code in memory, so debugging it more now should be relatively easy and painless...

vhelin, 2015-09-06:
Also please check with a hex editor that the output is correct. I'm done here for today...

bazzinotti, 2015-09-06:
just got your update.. Checking it out now

bazzinotti, 2015-09-06:
well my project doesn't compile anymore ahah
usually I would get
```
bazz@AwesomeSauce[~/Code/pce/game] wlapce main.s
Pass 1...
NUM_PCEBANKS = $24
Directive checks...
Internal pass 1...
DISCARD: ../engine/src/sprites.i: Discarding an empty section "zp-sprite".
DISCARD: ../engine/src/core.s: Discarding an empty section "core-code".
Internal pass 2...
Free space at $0000-$1ff5.
Free space at $2000-$2c92.
Free space at $3590-$47fff.
Bank 00 has 08182 bytes (99.88%) free.
Bank 01 has 05891 bytes (71.91%) free.
Bank 02 has 08192 bytes (100.0%) free.
Bank 03 has 08192 bytes (100.0%) free.
Bank 04 has 262144 bytes (100.0%) free.
292601 unused bytes of total 294912.
Total 294912 additional bytes (from headers and footers).
Total size 589824 bytes.
Free space at $0fd8-$1ff5.
Free space at $3590-$5fff.
Free space at $60e2-$7fff.
Free space at $a688-$47fff.
Bank 00 has 04126 bytes (50.37%) free.
Bank 01 has 02672 bytes (32.62%) free.
Bank 02 has 08192 bytes (100.0%) free.
Bank 03 has 07966 bytes (97.24%) free.
Bank 04 has 252280 bytes (96.24%) free.
275236 unused bytes of total 294912.
```
since I compile/link verbosely.
Now I just get
```
bazz@AwesomeSauce[~/Code/pce/game] wlapce main.s
Pass 1...
NUM_PCEBANKS = $24
LOAD_FILE_DATA: Could not open file "main.o".
rm: main.o: No such file or directory
```

I am able to assemble and link bug_exhibition issue 61 though.. I wonder
what happened.. :|

bazzinotti, 2015-09-06:
Also, I haven't even add the macro-macro call yet!

bazzinotti, 2015-09-06:
bug in pass_1? [since the compiler seems to just exit during pass_1]

bazzinotti, 2015-09-06:
I haven't made much change to the engine since the zip I provided you :3

bazzinotti, 2015-09-06:
pass_1() == FAILED confirmed.. I'm still tracing

bazzinotti, 2015-09-06:
pass_1 failure point 
```
pass_1.c:648
648	          return FAILED;
```

still tracing :)

bazzinotti, 2015-09-06:
this is part of 
```
if (q == INPUT_NUMBER_ADDRESS_LABEL)
          strcpy(mrt->argument_data[p]->string, label);
        else if (q == INPUT_NUMBER_STRING)
          strcpy(mrt->argument_data[p]->string, label);
        else if (q == INPUT_NUMBER_STACK)
          mrt->argument_data[p]->value = latest_stack;
        else if (q == SUCCEEDED)
          mrt->argument_data[p]->value = d;
        else
          return FAILED;
```

Do you make sense of this?

bazzinotti, 2015-09-06:
q == 0 for the failed check above. I'm not sure what else to do and I'm getting owned by the difficulty level.

bazzinotti, 2015-09-06:
the error occurs from the 3180th call to evaluate_token() -- counting from 1 .. still debugging..

bazzinotti, 2015-09-06:
I might have changed engine code enough that that 3180th call is not right for your version.. here go ahead and clone the updated repo: http://botb.club/~bazz/wladx-pce-engine.git

bazzinotti, 2015-09-06:
q == EVALUATE_TOKEN_NOT_IDENTIFIED sorry made a mistake earlier.. I'm not sure what the actual token is

bazzinotti, 2015-09-07:
I FOUND THE PROBLEM!!! Remember the work-around I created?!!! (search the page for "temporary work-around") You broke it!! 

```CALC_PCE_BANK_FROM_WLA_DATABANK \1 :\1```

That is now broken haha. Now I will try the original intended code (as in post1) and see if that at least works now.

bazzinotti, 2015-09-07:
Remember, the problem was identified NOT that the nested parameters wasn't working. IT WAS working already.. ONLY not for grabbing the bank byte..

I'm not sure what you implemented if it was even necessary :|

bazzinotti, 2015-09-07:
`CALC_PCE_BANK_FROM_WLA_DATABANK \1` is also broken.. 

The "token" that is failing "tmp" in 3180th call to evaluate_token()  is literally the string `CALC_PCE_BANK_FROM_WLA_DATABANK`

I hope this all helps you

bazzinotti, 2015-09-07:
I will try to make a bug_exhibition for this to simplify

bazzinotti, 2015-09-07:
Bug Exhibition Will simplify your life: http://botb.club/~bazz/wladx/issue59.zip

This used to compile just fine before your latest commit.

bazzinotti, 2015-09-07:
updated the zip -- pls re-download

bazzinotti, 2015-09-07:
I feel like I have to spell this out -- 
On a 1-level deep macro->macro call, recursive argument reference was working for me, except I could not grab the bank byte of the nested argument.. 

I'm not sure if your implementation somehow intended on extending whatever recursive ability may have been lacking.. but it's unclear to me what you've added, and how it broke the former functioning code.. 

And I re-stating this fact because I strongly feel you may have been thinking all macros were not able to recursively use arguments, since I've already proven they can to some degree. I do admit the title of this issue is misleading and maybe should be re-worded. 

I will write revert out of your commit and test how well wla dx can do macro argument recursion without your fix.. just as a test...

bazzinotti, 2015-09-07:
I am very displeased with how difficult it is to be clear tonight :[ Well I have to stop trying now and do other things.

bazzinotti, 2015-09-07:
http://botb.club/~bazz/wladx/issue59a.zip -- this compiles WITHOUT your latest commit just fine.. BUT, just like in this issue's original post, uncomment the BANK BYTE access and it fails...

I hope this enlightens you

cr1901, 2015-09-07:
Something to keep in mind: I receive an email for every comment you post. It's a bit annoying to come back from a day out to dozens of emails/issues to fix.

I'm a little bit confused that you're working on a project so complicated that you keep breaking WLA and finding new bugs. I do not deny that there are plenty of bugs to fix, but neither Ville nor I currently have time to fix all (in my case, any of) the bugs.

By all means, keep reporting bugs. But perhaps you should offer workarounds or use simpler macros for now. WLA's core does work, and for better or worse, there are multiple ways to accomplish the same thing. I've never had this much trouble with WLA, and I can't keep up with all the issues you're reporting.

bazzinotti, 2015-09-07:
Well I've made small bug_exhibition zip archives that prove to you. Look
for them.

bazzinotti, 2015-09-07:
@cr1901 http://botb.club/~bazz/wladx here is the entire cataloge of bug_exhibition archives.. Each is small enough :P

cr1901, 2015-09-07:
I should have some time this week to look at some bugs and use a debugger besides gdb. I'll take a shot at it then.

bazzinotti, 2015-09-07:
k... and I'll try to talk less haha

bazzinotti, 2015-09-07:
added one more bug exhibition (c) -- and here's a summary of all them: I've changed them all around so they make better sense now.

[A](http://botb.club/~bazz/wladx/issue59a.zip) - This shows the problem and what should work eventually
[B](http://botb.club/~bazz/wladx/issue59b.zip) - This is the temporary workaround.
[C](http://botb.club/~bazz/wladx/issue59c.zip) - This just shows that WLA can already do a recursive parameter 3 levels deep in macros already without ville's commit.

These were all compiled and tested without ville's latest commit, which I don't trust since heh since WLA already can handle recursive macros I'm not sure what he implemented. It seems the only fix necessary is to allow the bank byte to be accessed instead of segfaulting.. Maybe that's what he tried to fix???

bazzinotti, 2015-09-07:
This is a good update -- I hope this straightens everyone out on this Issue #59

[A](http://botb.club/~bazz/wladx/issue59a.zip) - This shows the problem and what should work eventually. It compiles now after ville's commit. I haven't yet verified it's correct output.

[B](http://botb.club/~bazz/wladx/issue59b.zip) - This was the temporary workaround. This broke after ville's commit, and it should not break. It's valid code.

[C](http://botb.club/~bazz/wladx/issue59c.zip) - This just shows that WLA can already do a recursive parameter 3 levels deep in macros already without ville's commit. but if you uncomment the line trying to access the bank byte, it fails (this is the issue itself again).

So we need to figure out the following things: 
1) Since C proves that WLA could do recursive-macro-argument-passing before Ville's patch, we see the fix for this issue is to just make the bank byte accessible.. Maybe that's what ville tried to do in his patch, but I'm concerned he did more than was necessary and I want him to acknowledge this concern. Either way, his patch broke prior functionality in B so we need to take action.

Either Ville will say "I just need to rework my patch" or he will say "Oh yeah I reimplemented that functionality and must redo the patch"

That's really all there is to it.

vhelin, 2015-09-07:
tl;dr but I'll see what went wrong...

bazzinotti, 2015-09-07:
that's good I was hoping you wouldn't read :) just get the bug_exhibitions
(A,B,C) there is a post that details all of them

vhelin, 2015-10-25:
Anyone tested the fix to this?

cr1901, 2015-11-15:
Will test within the next week; this week really isn't good (I say that as I perform maintenance to the Issues tracker).

vhelin, 2021-03-15:
I tested the problematic code today and it seems to assemble without crashes. Sorry that this took what 6.5 years? :D But I guess the bug was fixed earlier, no-one just rechecked this and closed the issue...

closed 2021-03-15 by vhelin