vhelin/wla-dx #60: Add ability to use environment-variables for file-includes

bazzinotti, 2015-09-06:
'nuff said... 
For now I just write my projects in a way that I can get around this desire.. ie from a central Makefile and every file is referenced from the directory the makefile is in.

But making projects with many object files and a central Makefile is sometimes a hard habit for me to get in when starting a new project -- it's during these times I wish I could use an environment variable to escape directory dependency...

Anyways -- this flip-flop attitude I have makes me wonder if this feature should even be added (worth the effort to add?) Let's talk about it ---> For one, environment variables are used differently whether on Windows and *nix -- but let's say WLA DX suddenly had the ability to handle environment variables -- do we have a way as App Developer to check the host OS so that we can conditionally include files in a platform independent manner? 

Food for thought...

Extra thought -- What platform independent manner can we include files to begin with using WLA DX (I remember learning this but I've since forgotten :X)

cr1901, 2015-09-06:
WLA requires a hosted ANSI C compiler, and any conforming C compiler will provide a getenv() that will either retrieve environment variables or return a NULL ptr IIRC. So platforms that support getenv() meaningfully get support.

> Extra thought -- What platform independent manner can we include files to begin with using WLA DX > (I remember learning this but I've since forgotten :X)
You can't. C makes no guarantees re: the format of paths that fopen and friends accept. I don't personally care for the .INCDIR directive and would much rather see them moved to the command line as parameters, similar to POSIX (and NOT an environment variable either). But that would require rewriting the command line parsing. And if we want to get really pedantic, hosted C compilers are not required to meaningfully support argc/argv[] :P.

Let's worry about the bugs first before we start adding new features.

vhelin, 2015-09-06:
pass_1.c hosts a function as follows:

int localize_path(char *path) {

  int i;

  if (path == NULL)
    return FAILED;

  for (i = 0; path[i] != 0; i++) {
#if defined(AMIGA) || defined(UNIX)
    /* '\' -> '/' */
    if (path[i] == '\\')
      path[i] = '/';
#else
    /* '/' -> '\' */
    if (path[i] == '/')
      path[i] = '\\';
#endif
  }

  return SUCCEEDED;
}

... and that is used with all the file include/incbin functions. So you can write the paths using ///'s or \\\'s, WLA will convert the path to the working format. Is this what you wanted?

bazzinotti, 2015-09-06:
Since this was a 2-part question, yes ville you answered the second part, whereas part I, which focuses on environment variables, is left non-conclusive until we finish bug-fixing, as per recommendation of cr1901.

galaxyhaxz, 2015-11-17:
I think a better way of doing this is making WLA accept a root include/library directory through the command line. This way we can specify  root files with <myfile> and local files with "myfile" just like in ANSI C. If you guys approve of this idea, i'll see if I can add it to wla myself sometime :) Example:

==> Makefile <==
AS = wla-65816
ASFLAGS = -o -d "../../include"
LD = wlalink
LDFLAGS = -vS -d "../../lib"

==> Main.s <==
.INCLUDE "myfile.h" ; searches current working directory
.INCLUDE <myfile.h> ; searches directory specified in the command line ("../../include")

==> Linkfile <==
[libraries]
bank 0 slot 1 "speed.lib" ; searches current working directory
bank 4 slot 2 <map_data.lib> ; searches "../../lib" as specified with parameter -d

vhelin, 2015-11-17:
> root files with and local files with "myfile" just like in ANSI C. If you
> .INCLUDE ; searches directory specified in the command line
> bank 4 slot 2 ; searches "../../lib" as specified with parameter -d

A bit off topic:

Sorry guys, I have your (also latest) emails about WLA DX, but currently
I'm busy with other things, and trying to get my life back on the track,
and very exhausted. But I'll check these later...

- Ville

galaxyhaxz, 2015-11-21:
Your time is always appreciated Ville. Anyhow I added this feature myself but I'm not sure how to upload it. Download it here and merge with the latest source:
https://www.mediafire.com/?7p7gr8o0z5d9imz
With this, you can use -I for wla and -L for wlalink to specify the include/library directory. This will be searched first before the current working directory. I have done testing and haven't had problems thus far, but more is appreciated.

closed 2021-03-15 by vhelin