vhelin/wla-dx #61: Should .INCBIN's FSIZE symbol require exporting to be visible in the same object file?

bazzinotti, 2015-09-06:
bug_exhibition : http://botb.club/~bazz/wladx/issue61.zip
```
.bank 0 slot 0
.org 0
.section "code"
	lda #<bin_size
.ends
.section "data1-gfx"
file_bin:
    .incbin "file.bin" FSIZE bin_size
    ;.export bin_size
.ends
```

Above will only work if the .EXPORT is uncommented. otherwise:
```
PARSE_STACK: Unresolved reference to "bin_size"
```

As I understand it, only symbols starting with '_' should be limited to section-scope -- I could understand it wrong.

vhelin, 2015-09-06:
Only labels that start with '_' are invisible outside their sections, this doesn't apply to e.g., .DEFINEs.

And yes, if you want to make a .DEFINE visible outside the file of origin, you'll need to .EXPORT it. Should apply to all internally generated .DEFINES like "bin_size"...

bazzinotti, 2015-09-06:
`And yes, if you want to make a .DEFINE visible outside the file of origin, you'll need to .EXPORT it. Should apply to all internally generated .DEFINES like "bin_size"...`

In my example, bin_size is not visible to the file of origin's other section. Is that expected behavior? That doesn't seem right to me.

vhelin, 2015-09-06:
Ah, but you make the definition later than you use it - if you want to make
this work, you'll need to create the definition lexically before using it.

bazzinotti, 2015-09-06:
OK thanks for setting me straight on that.

closed 2015-09-06 by vhelin