vhelin/wla-dx #62: Macro - a label passed as a named-argument has value $0000 in ROM

bazzinotti, 2015-09-07:
Doing things right this time :) -- Starting off with a bug_exhibition :dancer: 

`When using NAMED ARGUMENTS in a .MACRO, if the macro is called with a label as argument,
the named argument value is compiled/linked as $0000 as per the example below. You can see this
by viewing the rom file in hex (little-endian).`

http://botb.club/~bazz/wladx/issue62.zip

vhelin, 2016-01-31:
I just made wlalink to give more and detailed debug info when compiled with -DWLALINK_DEBUG, and wlalink revealed that only the computations on lines (of main.s) 27 and 28 are postponed to wlalink, apparently computations (that also give incorrect results) are resolved in wla -> need to fix wla as it's not able to know a label's address before the label has been placed into somewhere (in wlalink).

vhelin, 2016-01-31:
Ok, the reason for this bug was simple: in the code, there was an if that was supposed to handle this situation, but I just hadn't implemented it for some reason. Next to it was another empty if that was going to handle situations where you gave e.g. "label+1" to a macro, and referred to that macro argument using a named argument. If you used \1 etc. all worked, but the named argument handler was partially implemented.

I have no idea why I didn't complete the work (support for named arguments), perhaps I got distracted at the time of making it and when I came back I didn't remember that something was missing. :)

Anyway, I found another issue, and won't commit the fix to this until I've solved that new one.

closed 2016-02-01 by vhelin