vhelin/wla-dx #63: Incorrect symbols in .sdsctag causes silent failure

nicklausw, 2015-09-07:
`.sdsctag 0.1f,"New Mexico","","nicklausw"

Notice the "f" in the number. This makes wla-dx basically shut down silently somehow when it should make an error.

nicklausw, 2015-09-29:
Just a suggestion (I can't test right now,) but I think this is just a case of checking for isascii() or whatever the function is called inside input_number(). This might actually explain problems I've had before with WLA crashing out of nowhere.

vhelin, 2015-10-25:
This should be fixed now. Actually the integer/float parser was failing everywhere in the same situation, examples:

  10f
  100a
  1.056x

I don't know why I had left this one "open" in the sense that the number parser would exit properly and let the other parsers to continue from there, perhaps for situations like

  ld a, (10) ; "10" is followed by ")"

... but at least now situations like 1234A are spotted as errors when the parser expects an integer/float.

closed 2015-10-25 by vhelin