vhelin/wla-dx #65: INCDIR INCBIN cache bug

galaxyhaxz, 2015-10-20:
WLA caches files when used in INCBIN. This isn't a problem when using the following code:
DATA1: .INCBIN "world_1/level1.bin"
DATA2: .INCBIN "world_2/level1.bin"

However, if we use INCDIR to shorten this:
.INCDIR "world_1"
DATA1: .INCBIN "level1.bin"
.INCDIR "world_2"
DATA2: .INCBIN "level1.bin"

Will result in DATA2 being overwritten by DATA1, because of the same file name. I think this could be fixed by making WLA append the current INCDIR to the file name so that the whole file name will be used just like in the first example. I have been disassembling Super Mario World into a source code and I have noticed this is a frequent issue when using .INCBIN on many files. Hope this can be fixed.

vhelin, 2015-10-20:
> WLA caches files when used in INCBIN. This isn't a problem when using the
> following code:
> DATA1: .INCBIN "world_1/level1.bin"
> DATA2: .INCBIN "world_2/level1.bin"
>
> However, if we use INCDIR to [...]

Oops, what a silly bug. :) Thanks for the report, I'll try to fix it asap
(at the end of this week, perhaps)...

vhelin, 2015-10-24:
Hey, can you verify that the latest commit fixes this bug? :) I added a testcase for it that I used to check manually that the bug was fixed, but the more testing (especially by not the author of the bugfix) the better.

galaxyhaxz, 2015-10-25:
I tested with the latest build and it indeed worked. Thanks for the fix.

closed 2015-10-25 by galaxyhaxz

vhelin, 2015-10-25:
No prob! That bug has been there for years, LOL. :)