vhelin/wla-dx #67: .ELIF directive

cr1901, 2015-11-15:
I think a macro such as .ELIF/.ELSIF/.ELSEIF would come immensely in handy to avoid nested .IF/ELSE statements, or alternatively an input to the .ELSE directive which permits further .ELSE directives to follow, all terminated by a single .ENDIF.

Not sure how difficult this is to implement, but this is a wishlist feature for me, so I'll just leave it here for future reference!

vhelin, 2016-01-07:
This surely would be handy. As we already have .ELSE, adding support for .ELIF should not be that difficult, I think...

cr1901, 2016-06-03:
.ELIF should be identical to the corresponding .IF statements except for two differences:

* If the `ifdef` variable is zero, we are not inside a conditional, so error.
* Do not increment `ifdef` because we are not going into a new conditional scope.

Since so much code would be shared between .IF and .ELIF, the core logic of .IF probably should be spun out into its own file.

vhelin, 2016-06-04:
> .ELIF should be identical to the corresponding .IF statements except for
> two differences:
>
>    - If the ifdef variable is zero, we are not inside a conditional, so
>    error.
>    - Do not increment [...]

... or at least into its own function - not really sure about giving .IF
its own file as it's just one directive among many... :) But that'd be a
nice feature!

closed 2021-04-20 by vhelin