vhelin/wla-dx #69: Can't use literal arguments when using macro

Qix-, 2015-12-22:
I feel like I'm probably doing something wrong, but here goes. This is for commit 85291756463bbc3c3e9bcd0c7240cefaf46cafa6 (at the time of this writing, master/HEAD).

```asm
.MACRO P1_PRESSED ; (button) [A]
	lda \1
	and controller
	cmp #$0
.ENDM

.MACRO P1_PRESSED_A  ; (location) [A]
	P1_PRESSED #$80
.ENDM
```

fails with

```
STACK_CALCULATE: Syntax error. Invalid use of modulo.
```

But changing

```asm
	P1_PRESSED #$80
;       to
	P1_PRESSED 128
```

Allows it to compile.

Qix-, 2015-12-22:
Aha, I see. You have to do a literal *inside* the macro, i.e.

```asm
.MACRO P1_PRESSED ; (button) [A]
    lda #\1
    and controller
    cmp #$0
.ENDM

.MACRO P1_PRESSED_A  ; (location) [A]
    P1_PRESSED $80
.ENDM
```

Doesn't that make things slightly unreadable? If I don't know what `P1_PRESSED` does, then I would assume it's using `$80` as an address.

Also, what if I have a macro such as:

```asm
.MACRO FOO
    lda \1
.ENDM
```

And want to pass it either a literal, _or_ an address?

```asm
FOO $0100 ; Use an address
FOO #$0100 ; Use a literal (!! doesn't compile !!)
```

nicklausw, 2015-12-24:
Personally, I like that the literal is required to be in the macro. It helps me figure out what's going on inside.

The macro system can probably go without drastic changes to support literals on the outside. I think. Will need to look at how exactly macros are handled.

nicklausw, 2015-12-24:
This is...odd...what version of WLA are you using? My binaries aren't correctly using macros at all, they're reading all macro args as "unknown symbols".

Qix-, 2015-12-24:
Using wla-6502 v9.6 at commit 85291756463bbc3c3e9bcd0c7240cefaf46cafa6.

nicklausw, 2016-01-07:
Hey, check out the fix_65 branch of my wla-dx fork. Not sure if you're watching the repo, but if not then apparently there was already an issue open very similar to this one, and I forgot to mention this issue in the fix.

Kroc, 2020-01-22:
Bump. 6502/6510 assembling is essentially broken for me with this bug.

```
.MACRO  _set
        lda \1 ; does the # go here?
        sta pl_stack, x
.ENDM

.DEF SOME_CONST 1 ; or here?
_set SOME_CONST ; or here?
```

None of the options work. It's not possible to `LDA #` this way.

Kroc, 2020-01-23:
I'm thinking half of this problem is parsing of `#` as modulo operator.
Resolving the Macro issues may be made an easier task if modulo is changed:

* Use `%` (with whitespace either side) like most languages
* Use `//`, like some languages
* Use `.MOD` or `.REM` directives for absolute clarity

vhelin, 2020-01-23:
> I'm thinking half of this problem is parsing of `#` as modulo operator.
> Resolving the Macro issues may be made an easier task if modulo is changed:
> 
>     * Use `%` (with whitespace either side) like most languages

Currently that's reserved for binary values. People would also use it without whitespace and complain that it doesn't work properly...

> 
>     * Use `//`, like some languages

Currently that's reserved for comments...

> 
>     * Use `.MOD` or `.REM` directives for absolute clarity

Directives don't really work inside calculations... ???

# that a mnemonic uses is actually coded into the mnemonics struct so that's handled with a plain text comparison function. After # in the mnemonic comes usually a number or a calculation so that is handled by another parser that can handle calculations and numbers and labels and macro arguments. Moving the # handling from the mnemonic parser to the calculations/numbers/macro argument parser would make # a modulo, I think... I could always try to make it so that if a calculation starts with a #, then it's ignored, but then I would need to handle that somehow in the mnemonics parser... Complicated.

Kroc, 2020-01-23:
Operators like "+" and "-" are just strings, right? So can't the expression parser recognise ".MOD" instead of the directive parser?

vhelin, 2020-01-23:
Well, when we are looking for a number/address we scan through the current line and if we find '-' or '+' we'll give the control to calculation parser. That could also recognise ".MOD", but uh:

   LDA 10 .MOD 20

looks a little bit weird. :) A single character symbol would be much better...

Kroc, 2020-01-23:
In some languages `\` (rather than `/`) is used as an integer divide. That could work as a symbol.

Kroc, 2020-01-26:
Using `#` for modulo also creates difficulties with creating syntax highlighting definitions, like for VSCode!

vhelin, 2021-03-17:
Yeah, that # is part of the text in the mnemonic. We might somehow rip that (leading) # from the stack calculator, but that makes things difficult. Currently there are no easy solutions to fix this issue.

vhelin, 2021-04-28:
Commit 534342952f5154cdbec548819a87abe5dce980c7 made it so that now you can prefix whatever you feed a .MACRO as an argument with a '#', the system now eats that away and you can now ask  .if (\?1 = ARG_IMMEDIATE) inside the .MACRO.

That fixed GitHub issue #50 and should also fix this issue, but you'll need to do this now:

.MACRO P1_PRESSED ; (button) [A]
    lda #\1
    and controller
    cmp #$0
.ENDM

.MACRO P1_PRESSED_A  ; (location) [A]
    P1_PRESSED #$80
.ENDM

This is not going to change, so I'm closing this issue.

closed 2021-04-28 by vhelin