vhelin/wla-dx #75: Suggestion: maybe remove wla-6510

nicklausw, 2016-01-07:
The 6502 and 6510 processors have pretty much no differences, so I think WLA might just be better off getting rid of wla-6510. You can use wla-6502 to get pretty much the exact same functionality, there's just two different names for the program.

This is a drastic removal though, so if this is too much trouble, then I think decode_6510 and 6510_opcodes can be gotten rid of in place of 6502 at the least.

vhelin, 2016-01-16:
What do other people think about this removal of 6510? Any comments? :)

cr1901, 2016-01-16:
Far as I can tell, all the features of the 6510 can be implemented using a 6502 .MEMORYMAP; care must be taken to ensure addresses $0000 and $0001 are preserved, though (on-die I/O port).

nicklausw, 2017-05-22:
I'd still like this to be in consideration, because as far as I can tell wla-6510 doesn't actually have any extra functionality that wla-6502 lacks. Maybe more variables are managed but no new errors or output to object files.

vhelin, 2017-05-22:
If one can use wla-6502 instead of wla-6510, to create programs for 6510,
then I'm ok with removing wla-6510. Any volunteers for this task? :)

closed 2022-03-12 by vhelin