vhelin/wla-dx #79: wlalink crashes

hang-on, 2016-01-16:
[linkerbug.zip](https://github.com/vhelin/wla-dx/files/92824/linkerbug.zip)
I'm on the newest version of wla-dx. Wlalink crashes when I link objects and libraries. I have attached a minimal project that demonstrates this error. I have an older (before GitHub) version of wla-dx, and if I use the wla-z80 and wlalink from this version on the same files, no error, and everything links just fine.

vhelin, 2016-01-16:
Thanks, I'll try to check this one out asap.

cr1901, 2016-01-16:
@hang-on What command line args did you pass to the linker? This may be a duplicate of #55 and #24.

hang-on, 2016-01-16:
Thanks! This is what I write in the terminal:
wlalink linkfile.txt linkerbug.sms

I'm on Windows 10.

Files were compiled like this:
wla-z80 -o main.asm main.o
wla-z80 -l library.asm library.lib

vhelin, 2016-01-31:
Ok, I gave this a look with my 9-Jan-2016 (84a1a42ab0d367138258614a2b621f35fc5f0c79 [84a1a42]) sources of WLA DX - no crash.

Could you run wlalink under a debugger (before that e.g., edit the CMakeLists.txt, at the beginning of the file there are in comments how to build a debuggable version of the exes) and tell back what it says about the crash? :)

Gekkio, 2017-01-05:
I'm pretty sure this is caused by a stack buffer overflow and I've opened a pull request with a fix :smiley:

wlalink uses `fgets` to read the linkfile one line at a time, but assumes that every line has a trailing newline character. If this isn't true, it doesn't handle the EOF case correctly and there's a buffer overflow with the stack-allocated buffer.

Example linkfile (imagine the file ends at `.o` without a newline):

```
[objects]
danger_zone.o
```

During the first line the buffer will contain: `[objects]\n\0`
During the second line the buffer will contain: `danger_zone.o\0`

Now, look at the code at [files.c line 45](https://github.com/vhelin/wla-dx/blob/d434821f5b43a33f1450a315a8ac019aabaf4442/wlalink/files.c#L45). It attempts to trim newlines from the right side of the buffer but doesn't check for the null byte, so during the second line it continues reading past the buffer data, and can read past the stack-allocated `tmp[1024]` buffer and even end up writing a null byte outside it.

The example project attached to this issue doesn't have a final trailing newline character in the linkfile, so I'm 95% sure this is the issue here.

vhelin, 2017-01-06:
Thanks Gekkio for a very thorough work. 👍 :)

PR #133 (2017-01-06): Check for null byte when trimming linkfile line
If EOF happens before fgets reaches a newline, the resulting buffer
shouldn't be read past the automatically added null byte.

Most likely fixes #79
commit 55c8fd6b (2017-01-06): Check for null byte when trimming linkfile line

If EOF happens before fgets reaches a newline, the resulting buffer
shouldn't be read past the automatically added null byte.

Most likely fixes #79
closed 2017-01-06 by vhelin