vhelin/wla-dx #83: Hashmaps

Drenn1, 2016-01-29:
I want to improve wla-dx's performance. Again I'm dealing with a big rom, there are a ton of labels and I think linear searches of defines are the biggest culprit of slowness. Hashmaps seem like they'd be the best way to improve performance. Unfortunately C doesn't have a built-in hashmap structure.

With some googling, [libjudy](http://judy.sourceforge.net/) appears to be the most recommended implementation. But maybe you'd be uncomfortable adding such a dependency. I'll probably do something like this whether you pull it or not, but obviously I'd prefer if you pulled it. So basically I'm asking, how would you prefer I implemented this?

vhelin, 2016-01-29:
WLA DX has to compile on all the supported platforms, so libjudy would have
to work even on Amiga OS. :) Why not write your own, simple hashmapper
(even a trivial hashmapper algo would be better than linear search), or try
to locate one (coded in ANSI C) in the Internet... I'm usually against
reinventing the wheel, but not here. :)

I quickly checked and it seems that libjudy contains more core than WLA DX,
and we don't need a super general solution, just a couple of functions
here. :D

How about something like this:

  https://github.com/petewarden/c_hashmap

Google gives a lot of results for "simple hashmap c"...

Drenn1, 2016-01-29:
I'm just a bit cautious about taking some random person's code off the internet, and I don't really trust myself to write an original, efficient hashmap :P 

But ok, I'll use what you linked or something similar. I see a few issues listed so I'll see about testing it.

vhelin, 2016-01-29:
Oh course one has to read through the selected code, and test it... :) But
a simple hashmapper for WLA DX should not take more than a hundred lines of
code, I guesstimate... The hashmapper we need here doesn't need to be the
best hashmapper ever, even a trivial one should be enough for our needs.

cr1901, 2016-01-31:
>Again I'm dealing with a big rom, there are a ton of labels and I think linear searches of defines are the biggest culprit of slowness. 
We should probably add a profile target to CMake when compiling with GCC (to use gprof). In any case, I would like to know for sure if that is the cause of the slowness before optimizing.

Idea: instead of using a hashmap library, we could also possibly implement a trie using the bsearch() stdlib function?

Drenn1, 2016-01-31:
Not a bad idea, but I'm nearly done with hashmaps anyway - just making sure the darn thing works now. (There are too many features! :P ) My project's assembling more than 5x faster, which will certainly help my crummy laptop which took minutes to assemble.

cr1901, 2016-01-31:
WLA still builds on platfoms w/ 16-bit `int` (mainly b/c there's some utility to using tools only available for Amiga and DOS, and WLA let's you develop on these platform as well), and the code is C89. Please make sure your code at least compiles with `-ansi -pedantic-errors -Wall -Wextra` without problems.

closed 2016-02-05 by Drenn1