vhelin/wla-dx #85: Can't use macros unless indented

Drenn1, 2016-02-05:
Macros don't work when there's no whitespace preceding them on the line. Not sure if this is intentional or not.

I see in pass_1 that the possibility of it being a macro is entirely ignored when newline_beginning is set:

```
      /* is it a macro? */
      if (q == ss)
        m = macro_get(tmp);

      /* it is a label after all? */
      if (q != ss || newline_beginning == ON) {
```
Changing the label check, however, results in a variety of interesting errors.
```
      if (m == NULL && (q != ss || newline_beginning == ON)) {
```

I'm not sure why, but that change seems to make certain labels be interpreted as macros, usually the last macro that was used.

vhelin, 2016-02-06:
Yes, well, everything that's placed right at the beginning of a new line is either a ';' comment, a label or an assembler directive (this is what I remember without looking at any documents). :) This is an old formatting rule that I guess I got from some of the assemblers I used back in the day...

I don't think the world would lose anything if we modified WLA to accept macro calls right at the beginning of a new line, but as a task it's going to the "new feature" category. :)

Drenn1, 2016-02-09:
I see, so this is some sort of convention I wasn't aware of. My teacher mentioned this in my class today for the m68000, heh. Well no big deal, I just like to use macros like they're directives sometimes.

closed 2021-03-15 by vhelin