vhelin/wla-dx #97: wlalink: cmd arg for silent discard sections

nicklausw, 2016-03-30:
wlalink's ability to shed unused sections is good, but I don't necessarily want to be reassured every single build that sections are being discarded. I'm thinking "n" can be a command line argument that discards sections, but quiets down on telling you what is being discarded.

vhelin, 2017-05-17:
The unreferenced section discarder now obeys the -v flag of the linker, which should take care of this, I hope. :)

closed 2017-05-17 by vhelin

sverx, 2022-08-04:
I just stumbled upon this, I didn't realize it had changed.

Honestly I would prefer to have those messages back, but I also see the point in obeying the -v flag... so I wonder if it's possible to send the discard warning messages to _stderr_ in any case, even when `-v` is not specified?

What I mean is that I would love to get the `DISCARD: <section>` messages back but I can't usually be bothered with the other verbose info.

I could also dig a different verbosity level where discard messages are printed and then there's one (or two) line(s) summary at the end such as 
```
ROM: 174339 bytes (66.51%) free of total 262144.
RAM: 2415 bytes (29.48%) free of total 8192.
```
which could be set by a different parameter such as `-v0` or whatever makes more sense... :man_shrugging:

vhelin, 2022-08-07:
That -v0 things sounds good:

-v0:
Show DISCARD messages

-v1:
Show DISCARD messages
Show ROM & RAM summaries

-v2:
The current -v

?

sverx, 2022-08-12:
I just got this message even if I'm not using `-v` at all, so there is at least one message (even though it's in wla-z80) that didn't obey the verbose flag.

`DISCARD: HUD.inc: Discarding an empty section "HUD_update".`

vhelin, 2022-08-12:
Thanks for the info! I'll try to take care of this issue this weekend...

vhelin, 2022-08-13:
> I just got this message even if I'm not using `-v` at all, so there is at least one message (even though it's in wla-z80) that didn't obey the verbose flag.
> 
> `DISCARD: HUD.inc: Discarding an empty section "HUD_update".`

Ok, the latest commit should also take care of this.

sverx, 2022-08-17:
It seems to work. Which means that if you want the discard section messages to be printed you now need to turn on verbose, which will print lots of other info too. I guess we need -v1 (and maybe -v2) here too. Not that fundamental, though.