vhelin/wla-dx #99: [parser/Struct] Unnamed INSTANCEOF

bazzinotti, 2016-04-15:
I desire this functionality when I am extending a base structure, without losing the way the base structure's variables are accessed in the rest of the code. For example,

```
.STRUCT xpmp_fm_channel_t
dataPtr		dw	; 0
dataPos		dw	; 2
.ENDST

.STRUCT xpmp_fm_rhythm_channel_t
fm_chan INSTANCEOF xpmp_fm_channel_t
extra1	db
extra2	db
.ENDST

.ENUM $0000
chan1 INSTANCEOF xpmp_fm_channel_t
chan2 INSTANCEOF xpmp_fm_channel_t
chan3 INSTANCEOF xpmp_fm_rhythm_channel_t
.ENDE
```

The problem is that chan3 needs to access the base structure directly `chan3.dataPtr` but WLA DX does not allow this, it will only allow `chan3.fm_chan.dataPtr`.

That's why I say I want to perform an "unnamed" INSTANCEOF, which I do not believe is currently supported in WLA-DX.

# TLDR;

```
.STRUCT xpmp_fm_channel_t
dataPtr		dw	; 0
dataPos		dw	; 2
.ENDST
```
In the code, there are instances of the structure:

```
.ENUM $0000
chan1 INSTANCEOF xpmp_fm_channel_t
chan2 INSTANCEOF xpmp_fm_channel_t
chan3 INSTANCEOF xpmp_fm_channel_t
.ENDE
```

And the code references it, perhaps via macros, like so

```
.MACRO INIT_DATA_PTR
	; trimmed
	ld	(chan\1.dataPtr+1),a
.ENDM
```

---

Now, let's say through the development process, it is realized that an extension is to be made for channel 3. (This is the feature request)

First, what doesn't work: 
```
.STRUCT xpmp_fm_rhythm_channel_t
fm_chan INSTANCEOF xpmp_fm_channel_t
extra1	db
extra2	db
.ENDST

.ENUM $0000
chan1 INSTANCEOF xpmp_fm_channel_t
chan2 INSTANCEOF xpmp_fm_channel_t
chan3 INSTANCEOF xpmp_fm_rhythm_channel_t
.ENDE
```

the problem with this is that base structure can no longer be referenced in the style the rest of the code expects it

```
ld a,chan.fm_chan.transpose
```

That won't agree with the design of the entire code base though. This would:

````
ld a,chan3.transpose
```

but this won't work until somehow, structs can "namelessly" be declared in new struct definitions.

nicklausw, 2016-04-15:
I'm thinking the parser can be modified to do this:
```
.STRUCT xpmp_fm_rhythm_channel_t
INSTANCEOF xpmp_fm_channel_t ; notice no name
extra1  db
extra2  db
.ENDST
```

Depending on how structs are stored, this shouldn't be difficult to implement.

bazzinotti, 2016-04-15:
I've tried in the past using `INSTANCEOF` without a label before it, it doesn't work. But I suppose that could be the desired syntax :smile:
```
wla-z80 -o main.s main.o
main.s:33: DIRECTIVE_ERROR: Unexpected symbol "xpmp_fm_channel_t" in .STRUCT.
main.s:33: ERROR: Couldn't parse "xpmp_fm_channel_t".
make: *** [main.o] Error 1
```

nicklausw, 2016-04-15:
Yeah, this should be simple. If there's no name and an immediate 'instanceof' inside the struct, then simple, just copy the values from the struct straight to the enum/ramsection rather than to the label inside the struct.

ssjason123, 2016-07-06:
I just checked in a little change to my branch for this under Pass1.c (last 2 changes).
https://github.com/ssjason123/wla-dx/blob/parserUpgrade/pass_1.c

The changes only affect STRUCT I think similar changes could be added to ENUM and RAMSECTION for similar functionality.

The changes should support multiple unnamed structs as long as the child definition names don't collide:
```
.STRUCT FIRST
	VALUE	DB
.ENDST

.STRUCT THIRD
	VALUE3	DB
.ENDST

.STRUCT SECOND
	INSTANCEOF FIRST
	VALUE2	DB
	INSTANCEOF THIRD
	VALUE4	DB
.ENDST
```

vhelin, 2021-03-21:
Sorry for taking so long fixing this issue, I hope the latest sources work well as I didn't write the latest code used in .STRUCT/.ENUM/.RAMSECTION parsing it took me a while to understand how it works...

closed 2021-04-26 by vhelin