vhelin/wla-dx #100: WLA-DX Architecture Crash Course

bazzinotti, 2016-04-16:
Would any pro maintainers of WLA-DX be interested in doing maybe live-stream or Skype and go through the essential architecture of WLA-DX? I'm interested in a general walk through of the assembly and linking execution paths, perhaps highlighting top-tier functions involved. I'd like to go as deep as anyone's willing.

IRC is not a bad second option. I hang out in #wla-dx on EFNet.

If nobody answers the calling, I might start live-streaming a "How TF does WLA-DX Work?" series

nicklausw, 2016-04-16:
Ville is on a trip through Asia or something like that so this is kind of out of the picture for him.

cr1901 isn't around much, and doesn't seem to have much time when he is.

Honestly I'm no expert on the architecture, but I seem to be the only one who's around right now. All I want to know is what you want to know about WLA. The IRC thing seems to be working? but I've never used it so I don't know.

vhelin, 2016-04-17:
Haha, yeah, sorry guys, I'm currently scuba diving in Philippines, and I'm
happy if I can read news from the Internet and reply to my emails - live
streaming a video no-go, even YouTube is really choppy...  If you ask me,
instead of making a live video or spill out the details in IRC, it'd be a
much better idea IMO just to write down the information into a .txt file
and put that into Github along with the sources and READMEs etc. where
everybody can find it easily...

This is a good idea, and I think I'm the best guy to do it, but yeah, the
timing could be better. :)

vhelin, 2023-02-21:
I'm not recording any videos about WLA DX, but I could write more texts about it. I personally dislike video tutorials unless they are about a tool/thing that has a nice GUI like Unity 3D the game engine... WLA DX does not have a GUI.