vhelin/wla-dx #102: SNESDev Bugs

bazzinotti, 2016-04-17:
I've heard word of various WLA-DX bugs mentioned over at [SNESDev Forums](http://forums.nesdev.com/viewforum.php?f=12). Lately, I've only read about "so-and-so found a few bugs and doesn't use WLA-DX anymore." But, nobody has really filed any bugs here for us. I'm making this an issue to remind myself, or one of us .. to do a search for "wla" or "wla-dx" "wla dx" "wladx" etc. on the forum and see if any bugs can be discovered from any of the resulting posts... and then officially report them here on Github.

I know at least one of these bugs is having to do with SPC700, I'm not sure the others. Particularly, the person mentioned who found some bugs is called `psychopathicteen`, and may be worth getting in touch with on the matter.

These bugs should be filed as separate issues on Github, not as responses to this issue, but can be referenced with this issue to form a connection.

vhelin, 2016-04-17:
Greetings from the Philippines! Nice scuba diving, but internet doesn't
work very well... In my current hotel wifi seems to work, though...

It would be nice to have these bug reports in Github. "psychopaticteen" is
a lousy nick, bad initial impression. :) If you can convert their problems
into Github issues - nice! I'll try to do the same, also...

>
> I've heard word of various WLA-DX bugs mentioned over at SNESDev Forums.
Lately, I've only read about "so-and-so found a few bugs and doesn't use
WLA-DX anymore." But, nobody has really filed any bugs here for us. I'm
making this an issue to remind myself, or one of us .. to do a search for
"wla" or "wla-dx" "wla dx" "wladx" etc. on the forum and see if any bugs
can be discovered from any of the resulting posts... and then officially
report them here on Github.
>
> I know at least one of these bugs is having to do with SPC700, I'm not
sure the others. Particularly, the person mentioned who found some bugs is
called psychopathicteen, and may be worth getting in touch with on the
matter.
>
> These bugs should be filed as separate issues on Github, not as responses
to this issue, but can be referenced with this issue to form a connection.

bazzinotti, 2016-04-19:
Ville, thank you for your time. Enjoy the Phillipines 👍

vhelin, 2016-04-19:
No prob, I'm actually getting bored to travelling here. :) Too hot, too
much same same everywhere... Well, around 50 days more to go. :)

vhelin, 2021-03-19:
It's 2021 now and people who use software found on GitHub report problems they've encountered here on GitHub (by adding a new issue). I'm not personally going to search any forums for random problems with WLA and possibly convert those experiences to GitHub issues. Please advice the people on SNESDev Forums and elsewhere to come here to report their issues if they've found some. My free time is limited and is better used to solve reported issues than search through random forums hoping to find something to fix. :)

closed 2021-03-19 by vhelin