vhelin/wla-dx #103: Let's redesign the assembler arguments!

nicklausw, 2016-04-17:
I think the assembler args could use a redo.

I'm thinking we go more linux style, and change things up to be from this:
`wla-z80 -vo main.s main.o`
to this:
`wla-z80 -v -o main.o main.s`

Now, I get the feeling that this will not be very well liked, with the sole reason being "why?"

Well, the advantage? Extra arguments can ask for more information, like say a file name or directory, without making the args look ugly. The disadvantage? Every project written for wla-dx ever gets broken.

So while this will mess things up for plenty of people, I don't think changing build scripts to conform is THAT much to ask, since they're the same but with a different format. But before I do any work on this, I want to make sure that it will even be accepted into the main branch.

So, thoughts?

bazzinotti, 2016-04-17:
I am big fan of typical *nix argument ordering. Namely, that flags with arguments have them come directly after the flag. If it's not clear, I am saying that I support this notion.

I don't mind changing my WLA-DX projects to support this more logical and standardized flag processing. The WLA-DX user base is small and not "prime-time."

However, it may be worth adding a message to the invocation of assemblers. Something like
```
The argument ordering has changed. from (example):
argv[0] -vo main.s main.o
to this:
argv[0] -v -o main.o main.s

Please Type OK to acknowledge: 
```

When the user acknowledges, a new config file could be written that establishes it is no longer necessary to print this warning message.

However, adding a "config file" increases the complexity of the program considerably, in that for each platform, a suitable location must be identified. It might not be worth the effort. 

Informing users another way outside of the program's execution would be easier.

vhelin, 2016-04-19:
>
> I am big fan of typical *nix argument ordering. Namely, that flags with
arguments have them come directly after the flag. If it's not clear, I am
saying that I support this notion.
>
> I don't mind changing my WLA-DX projects to support this more logical and
standardized flag processing. The WLA-DX user base is small and not
"prime-time."
>
> However, it may be worth adding a message to the invocation of
assemblers. Something like
>
> The argument ordering has changed. from (example):
> argv[0] -vo main.s main.o
> to this:
> argv[0] -v -o main.o main.s
>
> Please Type OK to acknowledge:
>
> When the user acknowledges, a new config file could be written that
establishes it is no longer necessary to print this warning message.
>
> However, adding a "config file" increases the complexity of the program
considerably, in that for each platform, a suitable location must be
identified. It might not be worth the effort.
>
> Informing users another way outside of the program's execution would be
easier.

I don't think the config file is a good way to go, also for the mentioned
reasons. But the suggested *nix style arguments is ok for me. Could this
change be made so that both old and new syntax would work (but only the new
would be documented)? By checking if the file exists, file type, ascii
(percentage, 90%+) vs binary... ? And in the case of being uncertain, just
halt the execution and tell the user about the change in arguments? I just
feel a bit uneasy about breaking old scripts...

nicklausw, 2016-04-19:
Yeah, this can be done. If either a source file is placed after `-o` or one flag with a bunch of args is used like `-vpankergo` (making those letters up obviously) then the old style is used. If unsure, say "in with the new!"

nicklausw, 2016-04-19:
Would it be okay for a message to be printed when the old args are used?

vhelin, 2016-04-20:
I think it would be ok...

nicklausw, 2016-05-22:
I'm just about finished making the new arg parser. Does -D currently need no space between the flag and definition? Because either the parser is about to get weird, or it will from now on.

nicklausw, 2016-05-22:
#109 is in progress to implement new args.

bazzinotti, 2016-05-22:
@nicklausw I can't speak for WLA DX but mainstream args processors do not require a space.
`-Dsym` is same as `-D sym`

nicklausw, 2016-05-23:
Personally, I'd like to keep things simple. The old args made WLA do this weird system of building a string character by character to form a definition and stuff; now it's just "here you go!".

nicklausw, 2016-05-23:
I'm saying space is required.

nicklausw, 2016-05-23:
Does anyone mind if I rename ighesohrgpieshgoehrgeihgoiehrogiheorighoerhgofidhgoihesfosidhoihgoseihrgoihdofihgoiehrgosihrg.s?

nicklausw, 2016-05-23:
Welp, I forgot about terminal defines having actual values. This is going to get interesting.

nicklausw, 2016-05-23:
Is examples/65816/linker_test supposed to not link properly? NAME is set to kullervo (an undefined symbol) and the linker fails because it can't figure out what NAME is.

vhelin, 2016-05-23:
I do; the name is long for the purpose of testing that WLA can handle long
filenames. Or then again, if you must, you can rename it, but the new name
must be at least as long as the old one. :)

Want to make it better? Add more letters to the file name...

>
> Does anyone mind if I rename
ighesohrgpieshgoehrgeihgoiehrogiheorighoerhgofidhgoihesfosidhoihgoseihrgoihdofihgoiehrgosihrg.s?

vhelin, 2016-05-23:
It would be more than great if WLA could work both ways there, with and
without spaces... :)

vhelin, 2016-05-23:
The end user cares about flexibily more than about how WLA is programmed to
parse cmd line arguments... Just sayin. :)

>
> Personally, I'd like to keep things simple. The old args made WLA do this
weird system of building a string character by character to form a
definition and stuff; now it's just "here you go!".

vhelin, 2016-05-23:
All the "projects" under examples dir should assemble and link. Something
must be broken...

>
> Is examples/65816/linker_test supposed to not link properly? NAME is set
to kullervo (an undefined symbol) and the linker fails because it can't
figure out what NAME is.

nicklausw, 2016-05-23:
Is can_you_tell_this_name_is_intended_to_test_the_capability_of_wla_to_handle_long_filenames_and_stufffffffff.s a long enough name? I'd at least like it to make sense.

I'll try putting kullervo in quotes. Older WLA can't seem to handle that example much better.

I know the user cares more about flexibility, it just makes more sense to keep everything simple on the coding side. It isn't really that much to ask for saying "please put a space after -D when you use it".

vhelin, 2016-05-23:
>
> Is
can_you_tell_this_name_is_intended_to_test_the_capability_of_wla_to_handle_long_filenames_and_stufffffffff.s
a long enough name? I'd at least like it to make sense.

That's good!

> I'll try putting kullervo in quotes. Older WLA can't seem to handle that
example much better.

Ok

> I know the user cares more about flexibility, it just makes more sense to
keep everything simple on the coding side. It isn't really that much to ask
for saying "please put a space after -D when you use it".

But the less we break/remove old functionality, the better. :)

vhelin, 2016-05-31:
I'll leave this open as having (be it legacy) support for e.g.

  -Dsym
  -Dsym=10

would be nice :)

nicklausw, 2016-05-31:
That can be done, but it'll probably be weird, so you might wanna be ready to fix stylistics again.

vhelin, 2016-05-31:
No worries, code style formatting is pretty easy though monotonic... :)

nicklausw, 2016-06-01:
Okay so the args parser seems to work fine in my opinion, but over at NESDev, people are encouraging the use of getopt_long. Windows doesn't have it and Amiga probably doesn't either, so it'd have to be a copy paste job from gnulib which shouldn't cause licensing issues but might make the codebase kinda ugly. Thoughts?

cr1901, 2016-06-01:
I have no interest in supporting long arguments; I do not see what they accomplish other than being a GNUism.

On the flip side, I think POSIX getopt, even a custom-rolled one, is a very good ideal. But the reality is many programs (`find` being a good example) don't religiously follow POSIX, and WLA has a backwards-compatibility motivation. So I didn't really have a problem adding multiple `-q` options.

nicklausw, 2016-06-01:
Took a look at getopt.c. Kind of a giant, and I don't think anyone here likes the idea of replacing existing code with an external file that's oversized and is only there to make things look a little neater. (Not even sure if it does a good job at that).

vhelin, 2016-06-03:
If you ask me: let's forget it. :) The gain would be minimal and extra code
needs extra maintenance... What we have now is much better than before, and
good enough IMHO.

>
> Took a look at getopt.c. Kind of a giant, and I don't think anyone here
likes the idea of replacing existing code with an external file that's
oversized and is only there to make things look a little neater. (Not even
sure if it does a good job at that).

vhelin, 2016-06-03:
If I'm contradicting myself, here's what I think:

- Add a couple of code lines to make things more flexible, or to implement
a new, small feature: yes please!
- Add (a big, new file with) 300+ new lines of code for the same: no
thanks. :)

Like with the hash function before: add external hash library to the
project, or code/copy-paste just the few lines that were really necessary.
External lib lost. :) In some ways it was reinventing the wheel, but here
the wheel had become a full Tesla car already...

>
> If you ask me: let's forget it. :) The gain would be minimal and extra
code needs extra maintenance... What we have now is much better than
before, and good enough IMHO.
>
> >
> > Took a look at getopt.c. Kind of a giant, and I don't think anyone here
likes the idea of replacing existing code with an external file that's
oversized and is only there to make things look a little neater. (Not even
sure if it does a good job at that).

nicklausw, 2016-06-03:
I'm probably going to eventually try a rewrite to make the args parser very easily support lack of space between flag and arg without repetitive code. That's really the only advantage of getopt that I know.

vhelin, 2016-06-03:
Well, take a look at the simplifications I did (in the repo). :) If you can
make the code even simpler - go for it!

>
> I'm probably going to eventually try a rewrite to make the args parser
very easily support lack of space between flag and arg without repetitive
code. That's really the only advantage of getopt that I know.

nicklausw, 2016-06-03:
Sweet! I must have missed that.

nicklausw, 2016-07-14:
*sneaks in* Closing this because main objectives were done *backs away into darkness*

closed 2016-07-14 by nicklausw