vhelin/wla-dx #105: Testing Framework

bazzinotti, 2016-04-19:
I'm not sure if WLA-DX has "got the ball rolling" toward having unit tests, but it's damn time to have a testing framework.

If anyone has an idea of key testing areas, or towards a proper approach... I'm listening.

bazzinotti, 2016-04-19:
For anyone who doesn't yet realize why a testing framework is great; It helps ensure that nothing breaks from new changes to code. At least, that's the point. And new tests are created for new features when they are added.

For something like WLA-DX it's important because sometimes a test to "module-A" could break something in "module-Z" unknowingly. The tests could possibly detect such an instance.

Unfortunately, tests are created by humans, so nothing is perfect.

cr1901, 2016-04-19:
I have a branch for this that utilizes ctest, but it's out of date. Also WLA currently isn't all that suited for unit testing. We have the test files and bug exhibition directories, but not functions which can be pulled out and tested individually.

bazzinotti, 2016-04-19:
I'm not experienced with cmake/ctest. Crap, another thing to learn.

> Also WLA currently isn't all that suited for unit testing

You do agree that it's time to "suit up," though, right? We need to
seriously start thinking of the functions we need to create tests for. If
we could have a clear idea of the kind of tests we need to create, that
would be a wonderful step in the right direction. And it's definitely
something I can try contributing to ATM without any ctest experience.

cr1901, 2016-04-19:
I would love to get ctest up and running for the existing test files, but I'm currently not able to work on WLA meaningfully.

vhelin, 2016-04-19:
Unit testing WLA DX (linker and especially the buggier assembler) would be
super nice! Our current tests really don't tell that much, and are far
outside the core code... But before that's possible a lot of huge functions
need to be split into smaller ones and made more independent. Perhaps one
at a time, one function gets refactored, then ctests for it, rinse and
repeat... I've never used ctest myself either, and can't promise to do much
in the following 50 days. Up to +35C every day here on the Philippines (and
later Thailand), it's even hard to remember to eat and drink enough. For
working I'd need an aircon room with working wifi, and that'd be luxury
(e.g. expensive) here... :)

>
> I would love to get ctest up and running for the existing test files, but
I'm currently not able to work on WLA meaningfully.

vhelin, 2021-03-19:
Ok, it's 2021 now, and I'm happy just writing tests for the assembled binaries. If anyone wants to start unit testing WLA's code then that's more than nice, but I'm targeting my efforts into fixing old bugs, adding new features and testing that what WLA assembles is and stays correct (using byte_tester.exe).

vhelin, 2022-08-29:
Getting back to this in late 2022: 

"For anyone who doesn't yet realize why a testing framework is great; It helps ensure that nothing breaks from new changes to code."

This is what our tests in the /tests folder test (by running run_tests.sh).

In my years as a software engineer (or software developer, I think that's a better title) I've found out that unit tests tend to test trivial things and they are really not that useful anywhere. Instead adding new tests to the /tests folder that test that the assembler and the linker really work as they should is the way to go, if you ask me. That is all that matters that they produce the expected output in the end.

And for that we have "duplicate" issue https://github.com/vhelin/wla-dx/issues/376 - thus I'm closing this issue.

closed 2022-08-29 by vhelin