vhelin/wla-dx #106: Docs organization

nicklausw, 2016-04-19:
Over at the NESDev forums that Bazz and I keep mentioning, people have complained about the README being difficult to navigate and understand.

I'm not entirely sure what to do about understanding, but something I've noticed is that the directives are out of order. Sounds painful to fix, because it is, but it's possible.

Something that I'd like to do that's a bit more long-term is maybe turn the readme into an HTML file. Why? Well, because hyperlinks. They will make it much easier to navigate the large file.

I should probably take a break from opening more issues and actually work at some of the existing ones, but for now this is just being thrown up in the air. I am a procrastinator, it's inevitable.

nicklausw, 2016-04-19:
Yeah, some areas of the README are outdated...Ville's contact info and address, WLAD stuff, supported OS's (there's not really a point since the program is source-only now), the "bugs" section should link to here rather than to an e-mail address, the "files" section doesn't describe nearly as many files as there are in the repo, just some gameboy examples.

nicklausw, 2016-04-19:
Also in section 3.6...
```
and #20.b
and #20.w
bit loop.b
bit loop.w
```
This isn't recommended, it doesn't seem to work well if you ever do arithmetic. I think this should be changed to...
```
and.b #20
and.w #20
bit.b loop
bit.w loop
```

bazzinotti, 2016-04-19:
>  it doesn't seem to work well if you ever do arithmetic

Can you please provide a bug_exhibition that demonstrates "not working well"?

When I was making contributions to [mitmproxy](https://github.com/mitmproxy/mitmproxy), I learned about [RST/Sphinx](http://www.sphinx-doc.org) for documentation. I find it really cool because the native text markup is well-readable on its own, but can also be used to render offline or online documentation. There is even a theming component which seems separate from the actual documentation, which I think is cool. Here is mitmproxy's [doc](http://docs.mitmproxy.org)

bazzinotti, 2016-04-19:
I think first we ought to
- restructure the README's organization
- fix grammar
- ?

before making it into something foreign

nicklausw, 2016-04-19:
By not working well, I don't mean a bug of some sort, I mean it just doesn't look good. I'd recommend making "lda.w #$40" rather than "lda #$40.w".

bazzinotti, 2016-04-21:
@vhelin is that functionally different syntax? @nicklausw my concern is that the two are not functionally identical.

vhelin, 2016-04-22:
I have to say that "lda.w" was a little bit bad choice as if you code e.g.
in 680x0 asm, "lda.w #$40" means another thing than "lda #$40.w", thus I'm
not a big fan of "lda.w" style myself... And about something "not looking
good", that's just in the eye of the beholder, and not really worth a good
enough reason to make changes. Functionality and logic first. :)

vhelin, 2016-04-22:
In WLA DX they mean the same thing...

nicklausw, 2016-04-22:
Does the README at least state that there's two different ways of doing it? Because other assemblers (kxas, bass) use lda.w, so the functionality should at least be documented.

bazzinotti, 2016-04-25:
Consider analyzing whether the info in `.8BIT` is valid: 

```
LSR 11       ; $46 $0B

LSR 11       ; $4E $0B $00
```

and whether the `BIT` directive has an effect for immediate values... If it doesn't .. it's confusing