vhelin/wla-dx #107: 16-bit Inference Observation

bazzinotti, 2016-04-25:
copy the `bug_exhibition/65816/template_project` directory and use the following code snippet as your `main.s`, then simply use `make`

Please read the headline blurb below

```

;»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»
; 65816 bug exhibition -- 16-bit Inference Observation
;
; This demonstrates that WLA-DX can already infer labels as 16-bit naturally,
; by the immediate addressing mode, and via JMP.
; However, not for the absolute addressing mode. What gives?
;»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»

.MEMORYMAP
DEFAULTSLOT 0
SLOTSIZE $2000
SLOT 0 $0000
SLOT 1 $2000
SLOT 2 $4000
SLOT 3 $6000
SLOT 4 $8000
.ENDME

.ROMBANKMAP
BANKSTOTAL 1
BANKSIZE $2000
BANKS 1
.ENDRO

;»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»
; main
;»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»

.DEF INFER

.bank 0 slot 4
.org 0
.section "derpface"

label:
  rep #$20
  lda #label    ; correctly inferred
.IFNDEF INFER
.16bit
.ENDIF
  lda label     ; "FIX_REFERENCES: Value ($8000) of "label" is too much to be a 8bit value."
  jmp label     ; correctly inferrred

.ends

```

bazzinotti, 2016-04-30:
so tell me, why can we do `jmp label` but not `lda label` ?

nicklausw, 2016-04-30:
There is no 8-bit jmp, so 16-bit becomes the default.

bazzinotti, 2016-04-30:
OK, and why does `lda #label` work?

nicklausw, 2016-04-30:
Not sure. Will try to find out next time I'm at a computer.

nicklausw, 2016-05-01:
https://github.com/vhelin/wla-dx/blob/master/opcodes_65816.c Odd...there is no `lda #16-bit-value`. There is, however, a `lda 16-bit-value`. Your example, when linked, uses 0xA9 for `lda #label` which is not the right opcode. I think it has to do with opcode typing; the linker thinks that with absolute addressing, the 8-bit kind is just the universal kind. With immediate addresing, however, it knows that there is a 16-bit version to be used.

I need to investigate how opcode types are carried into the linker, if at all. They might just be used by the assembler.

nicklausw, 2016-05-01:
Well wait, WLA appears to pick up on opcode size on its own. That's probably why the absolute addressing makes no error, it knows to be 16-bit.

nicklausw, 2016-05-01:
...And apparently WLA doesn't use the wrong opcode for absolute addressing, either. I'm stupid.

bazzinotti, 2016-05-02:
> Your example, when linked, uses 0xA9 for lda #label which is not the right opcode.

Actually, that's the right opcode.

bazzinotti, 2016-05-02:
Everybody, the point of the conversation is deduct why immediate addressing of a label without hinting works fine for 16-bit, but not absolute addressing (and likely the other non-immediate addressing modes as well), and to see if this ability can be transversed into the other addressing modes.

nicklausw, 2016-05-02:
As I said, opcode size tracking. WLA knows that a is 16-bit, so `lda #label` works because it assumes label is 16-bit. However, `lda label` could be anything.

bazzinotti, 2016-05-02:
Sounds right. That's much clearer