vhelin/wla-dx #108: Control characters in chars and strings

maxim-zhao, 2016-05-11:
I'd like to be able to type things like '\n' and "Hello, world!\n\0".

nicklausw, 2016-05-11:
I didn't actually know WLA couldn't do this already. Should be pretty simple to implement in string parser, might break some existing code but not enough to cause much a problem (example: if existing code uses `\n`, just use `\\n` instead).

Sorry for not working on this program much lately, the NESDev comp has come up and I'd like to participate in that, and SNES stuff too. I'll look into converting some of my current code to use WLA, to encourage me to try working on it some.

vhelin, 2016-05-12:
There should be some support for this in WLA DX, not sure why/if it's not
working. Will take a look later. I'm currently on Malapascua island in
Philippines scuba diving, and the internet doesn't work very well here...
Blackouts, limited supplies, etc... But a great place to scuba dive! :D

maxim-zhao, 2016-05-12:
`.db "TEST\r\n\0\t"`
Result:
`54 45 53 54 5c 72 5c 6e 5c 30 5c 74    TEST\r\n\0\t`
Same for `.asc`.

bazzinotti, 2016-05-12:
This used to work. commit 7e40b31785654956316ddb7b03aa802d605fdfea can do what you want. That is the commit I use for my personal development and I use strings with escape sequences all the time. I'm not sure what they've been adding to WLA-DX but I can't even assemble my game anymore with master branch .. I just use the commit I referenced :)

To WLA-DX devs, there is a pretty large gap of commits between the one I refer to, and the master. It is your job to see which commit causes the escape sequences to break!

bazzinotti, 2016-05-13:
I am wrong -- my printf driver actually parses '\' and then 'n' for implementing '\n' processing -- my apologies.

galaxyhaxz, 2016-07-05:
I am not 100% sure if it still works but I made a commit #77 that adds several ASCII commands. It adds "\n, \0, \x, \> and \<". The last two are to set the highest bit of the character before or after, and are temporary until a better way is made.

vhelin, 2021-04-28:
Pretty amazing that this issue hasn't been fixed already. :) Quickly reading the last comment before this made me think this was taken care of, but now that I test WLA v9.13a (28-Apr-2021) I see that \\r, \\n, \\t aren't taken care of... I'm fixing this next. :)

closed 2021-04-29 by vhelin