vhelin/wla-dx #112: wla-gb silent fails with non-opcode symbols?

nicklausw, 2016-06-01:
In the process of converting an RGBDS project to WLA...
```
include "src/gb.i"
```
WLA just stops right here.  Returns quietly in Cygwin and Command Prompt when I know this is an error.

nicklausw, 2016-06-01:
Does the hashmap thingy know to give an error when things don't work?

nicklausw, 2016-06-01:
Wait, not the hashmaps. Still can't figure this out.

nicklausw, 2016-06-01:
It has something to do with failure with loose quotes, which I thought was fixed before.

vhelin, 2017-05-18:
Do you still get this behavior? I tried to replicate the issue, but there isn't much info about it...

nicklausw, 2017-05-18:
That's the entirety of the issue. If your wla gives an error for a loose include then I guess it's been fixed. If it doesn't then something's up. I probably don't actually have the file where the error happened anymore.

nicklausw, 2017-05-18:
I should mention that I'm not at a computer right now, otherwise I'd test this myself.

vhelin, 2019-09-06:
I just tried to include a file that doesn't exist and WLA gives me a clear error about it. Do you still get a silent exit without any error messages in a similar situation in Cygwin (or elsewhere)?

I remember there was a bug that was fixed by adding a fflush() to directive_print() - the printed string didn't appear on screen in Cygwin - perhaps this is similar and we could fix this by adding fflush() to print_error()?

closed 2021-03-15 by vhelin