vhelin/wla-dx #116: Append to a Section?

Kroc, 2016-06-07:
Being able to append to an existing Section is critical for supporting 'extensible source'. Let's imagine that each level in a game requires data in five different data tables in five different source files. This is currently how people currently write their game and it's a huge hindrance to maintenance, flexibility and extending the source.

If we could append to a Section, we could instead define each level in its own file, where in the level would append it's particulars to each existing Section. Therefore levels could be included / excluded very easily without having to actually modify the source.

E.g.

```
.section someSection
    db 1, 2, 3
.ends

; in some other file somewhere
.append someSection
    db 4, 5, 6
.enda
```

nicklausw, 2016-06-08:
Not completely sure if this can be done or not, will have to check it out, but as a workaround...

```
.section "someSection"
  .db 1, 2, 3
end_someSection:
.ends

; in some other file somewhere
.org end_someSection
.section "more someSection" force
  .db 4, 5, 6
.ends
```
Inconvenient but seems to do what you want for now.

cr1901, 2016-06-08:
`.SEMIFREE` should do this already?

nicklausw, 2016-06-08:
From the looks of it, `.SEMIFREE` works if the original someSection was the last section declared in that bank. Otherwise things won't line up correctly.

Kroc, 2016-06-08:
An Append would need to implicitly reuse the metadata of the original Section; Bank/Slot &c.
E.g.

```
.bank 4
.section someSection
.ends

.bank 6
.append someSection
; this content goes in bank 4
.enda
```

cr1901, 2016-06-08:
Oh, you're asking for something similar to `GROUP` in MASM?

Kroc, 2016-06-14:
I can't say for sure because the terminology, semantics, and behaviour of assemblers varies so much.

Here's essentially what I'm looking for:

```
;assigns the label's address, but populates no data yet
TABLE levels

;elsewhere...
ROW .level1 TABLE levels {
    ;a label "levels.level1" is defined and this data appended to the table's contents
    BYTE 1, 2, 3 ...
}
```

We define a 'section', but begin with zero content. Each 'row' encountered elsewhere in the source inserts that data where the table originally occurs; as if it had been written there.

In my case, a level in a game might need to include small bits of data in twenty tables spread across the ROM; and to be able to keep that data beside the level in question would be a huge benefit as individual levels could simply be enabled or disabled without having to modify many disparate source files.

vhelin, 2017-05-19:
I added support for this, syntax is

.section "source"
.db 1
.ends

.section "bonus" appendto "source"
.db 2
.ends

I'm not 100% sure I've done all I need to do codewise for this to work, but the small testcase I wrote worked. If you guys could test this more, then thank you!

closed 2018-05-03 by vhelin

Drenn1, 2018-05-03:
I'll try this out when I have a minute, thanks :)