vhelin/wla-dx #117: [Just an idea] linkfile functionality to the command line

nicklausw, 2016-06-08:
I'm kinda afraid to work on the program with the huge commits from Jason hanging over, so I'd like to do some more arg work at the least.

Linkfiles are not super convenient. Added a new file to the source tree? Well too bad you forgot to add the object to the linkfile so wlalink is freaking out over missing code or something. Just adds another layer of complication and subtracts one of flexibility.

Here's my idea to not remove linkfiles but make them unnecessary.

Let's say we have a linkfile as such:
```
[objects]
main.o

[libraries]
bank 0 slot 2 base $80 idunno.l
```

wlalink would be called like this (assuming the linkfile is named, of course, "linkfile"):
```
wlalink linkfile outfile.out
```

What could happen is that the args are changed to move the linkfile out of the way...
```
wlalink -lnk linkfile -o outfile.out
```
...and who says we need a linkfile at all?
```
wlalink -o outfile.out main.o idunno.l
```

The problem that arises is that the extra info for libraries won't go inside the command line very well. So let's just allow it in the original source file instead.
```
; in the library source...
.bank 0 slot 1
.base $80
```

The library format would be changed to be able to tell the bank, slot and base if inside source, and wlalink would be changed to decipher between given object files and library files.

Headers, footers and definitions can fit easily.
```
-hdr header.dat -ftr footer.dat -D ahhh=2
```

This will be more difficult to implement than the plain args were. Thoughts?

ssjason123, 2016-06-08:
I really like the idea of getting rid of the linkfile. 

Libraries really pose a problem though. I am not a fan of including the library location directly within the library itself, unless it can be overridden somehow. I think hard coding the location within the library takes away a big part of what makes a library unique over a standard object file, that it can be dynamically relocated and reused between projects. I haven't actually used a library on my projects so I could be completely wrong there though :).

nicklausw, 2016-06-08:
Perhaps libraries can be individually added like...
```
-l library.lib 0 2 $80
```
...with the base being optional? Neither solutions work (or look) super well but the way WLA works, there isn't really much of a better way.

nicklausw, 2016-06-09:
I should mention that over at SMSPower, many compression "libraries" such as aPLib and PSGlib are not distributed as libraries; they already assume you'll change the bank and slot within the source yourself, and are just regular source files.

maxim-zhao, 2016-06-09:
How about special-casing the single-source-file case? That covers the majority of projects I've worked on. The only reasons I can see for linking multiple files are
* Attempted closed source libraries
* Saving compile time

...and neither of these is a very compelling reason to maintain a linkfile. Personally, I `.include` everything and that's perfectly fine, "compile" times are near-instant on even the oldest of computers.

More exciting to me would be a single-invocation assemble+link, even if internally it is two-phase.

nicklausw, 2016-06-09:
What do you mean by special casing? Not much reason for a linker phase if one object file is always going to be used regardless. Although my Sudoku game is probably WLA's only usage of multiple modules...

Also are you asking for a program that calls upon the assembler and linker in one? That could be done.

maxim-zhao, 2016-06-09:
I think i was a bit confused :smile: There's no real difference between 1 `.o` and _n_ `.o`s, as in your original post.

galaxyhaxz, 2016-07-05:
Why not just make it more like ANSI-C and get rid of the linkfile altogether? I would much rather specify objects in my makefile, and in turn, just like in C/C++, specify libraries. An example could potentially look like this in a makefile:

LIBDIR = $(SDK_HOME)/libs
LDLIBS = -lstrings -ldebug -lmapcreator

-------- WOULD BECOME -------

LDLIBS = -lstrings 0 2 $80 -ldebug 0 1 $80
LDLIBS += -lLevelTester 0 3 $80
LD = wlalink

all:
$(LD) $(LDLIBS) -o main.o testlevel.o

Drenn1, 2016-07-05:
fwiw, it's not hard to generate linkfiles in a makefile, so you never need to touch it. Although this particular recipe is unix-based.

```
$(TARGET): $(OBJS) linkfile
	$(LD) -S linkfile rom.gbc

linkfile: $(OBJS)
	@echo "[objects]" > linkfile
	@echo "$(OBJS)" | sed 's/ /\n/g' >> linkfile
```

That said, passing the objects through the commandline would get rid of the middleman here.

AntonioND, 2019-12-08:
Yes, please, this feature would be really useful... Right now you need to duplicate the list of files between your build system and the linkfile, or generate the linkfile from the build system, which is awkward.