vhelin/wla-dx #118: CADDR not working properly in macros

Drenn1, 2016-07-05:
CADDR is behaving oddly when used in macros. It seems to get messed up when the argument uses arithmetic. It seems to affect the parserUpgrade branch also.

This main.s file was based off of the gb_z80 bug template.

```
;»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»
; An example showing weirdness with the use of CADDR in macros.
;»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»

.MEMORYMAP
DEFAULTSLOT 0
SLOTSIZE $2000
SLOT 0 $0000
SLOT 1 $2000
SLOT 2 $4000
SLOT 3 $6000
SLOT 4 $A000
.ENDME

.ROMBANKMAP
BANKSTOTAL 4
BANKSIZE $2000
BANKS 4
.ENDRO

;»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»
; main
;»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»»

; Big endian .dw macro
.macro dwbe
	.dw (((\1)&$ff00)>>8) | (((\1)&$ff)<<8)
.endm

; define lower byte, ignore upper
.macro dblower
	.db (\1)&$ff
.endm

.macro dbplusone
	.db (\1&$ff)+1
.endm

.ORGA $0150

label:
	; Should be $150-$150=0, instead is $150
	dwbe label-CADDR
	; Should be $152-$150=2, instead is $feb0 (-$150)
	dwbe CADDR-label
	; Should be $154+1=$155, instead is $0001
	dwbe CADDR+1
	; Should be $57, instead is $01
	dblower CADDR+1

.ORGA $0160

label2:
	; The following work properly (they should be equivalent to using the
	; macros, but aren't)
	.dw (((label2-CADDR)&$ff00)>>8) | (((label2-CADDR)&$ff)<<8)
	.dw (((CADDR-label2)&$ff00)>>8) | (((CADDR-label2)&$ff)<<8)
	.dw (((CADDR+1)&$ff00)>>8) | (((CADDR+1)&$ff)<<8)
	.db (CADDR+1)&$ff

.ORGA $0170

	; This works correctly, because the arithmetic isn't passed to the
	; macro?
	; Result is $71, as expected
	dbplusone CADDR
```

ssjason123, 2016-07-06:
I tried running the code through a fresh code from the parserUpgrade branch but I wasn't able to replicate the issue. It gave me the expected values when I loaded the ROM up in BGB. 

The parserUpgrade branch should consume the entire macro argument up to the delimiter and substitute it directly in \1-n where the old branch would evaluate it prior to substitution so it should be treated identically to the label2 samples as the tokens are processed. 

I don't have any good suggestions, just to double check that wlalink is up to date since its where the CADDR is evaluated at.

Drenn1, 2016-07-06:
...Well, my problem was that I forgot to rebuild main.o when I switched branches. It redid the linking, but I didn't notice that it wasn't redoing the compilation. It would seem that wla-gb was the problem, not wlalink, though that doesn't really matter. Sorry for wasting your time. :)

closed 2016-07-06 by Drenn1

ssjason123, 2016-07-06:
It's all good. Glad you got it working :).