vhelin/wla-dx #125: Non unix-Eol causes library linking issues

galaxyhaxz, 2016-09-29:
So this bug took me forever to find. I made a library and after some changes it all the sudden failed to link. The library will compile fine, but spit out random errors when wlalink links it to my project. After lots of trial and error, it ended up being the EOL that caused the issue. All my files have a Windows EOL (0x0D+0x0A). If I do an EOL conversion to Unix (0x0A only) the library links just fine. I'm currently looking at the issue myself, it seems somewhere WLA fucks up the line count or something because most EOL parsing uses 0x0A only. This however skips a byte and messes up the linker's address to code in the library. Causing it to read code in the wrong spot. Anyone had this issue before?

vhelin, 2016-09-30:
Urgh!

The preprocessor should change all 0xD+0xA combos just to 0xA thus there is
no need to check for 0xD later... If the preprocessor is leaky, then it
should be fixed... :P

vhelin, 2019-05-23:
On line 685 of include_file.c (inside preprocess_file()) you can see that 0x0D is just skipped...

I'm still interested in solving this issue. Can you repeat it with the latest sources? Or can we close this one?

galaxyhaxz, 2019-05-23:
Heh sorry for the 3 year late reply haha. I'm not working on the fork anyone so you can feel free to close. I don't even remember much about it lol.

vhelin, 2019-05-28:
I converted a few test projects to DOS format, also tried them in UNIX format, but all seems to work... I'll close this, but if something similar comes up, open a new issue here.

closed 2019-05-28 by vhelin