vhelin/wla-dx #127: Expressions should be allowed to be out of 16-bit range

maxim-zhao, 2016-10-13:
In some places it makes sense to use numbers greater than $ffff, for example:
`.unbackground $4c000 $7ffff`
If I use an expression then it doesn't work:
`.unbackground 19*$4000 $7ffff`
`STACK_CALCULATE: Out of 16-bit range.`
I think the calculation should be unbounded (within reason) and then only bounds-checked once it is clear what the width needs to be (whether that's 16, 8 or some other number of bits).

vhelin, 2017-05-14:
Sounds good. I'll try to check out how much work this would require in the coming days...

vhelin, 2017-05-14:
In fact it looked like the boundary check in stack_calculate() was unnecessary as the result size is checked outside that, where the results are used. There might be cases where the check is not done outside, but I couldn't find any, though I made just a quick check around the code. Let's assume the problem is solved.

closed 2017-05-14 by vhelin