vhelin/wla-dx #128: Z80: SUB ixh/ixl not recognised

iamgreaser, 2016-10-21:
Pretty easy to test: try to add `SUB ixh` to your code.

`SUB iyh`, on the other hand, works perfectly.
`AND ixh` also works fine as well...

maxim-zhao, 2017-01-10:
They are listed as `SUB a, ixh` and `SUB a, ixl`: https://github.com/vhelin/wla-dx/blob/master/opcodes_z80.c#L1151 but I'm not sure if it's as simple as editing that file?

vhelin, 2017-01-10:
Yep, it should be as simple as that;

Just edit opcodes_z80.c, make sure the bytes (type etc.) are correct, check
out other similar opcodes and the definition of "opcode" in defines.h. When
you run "make" it will generate the opcodes_z80_tables.c speedup lut, and
everything should work.

As I'm no Z80 user I really don't know how correct/incorrect the opcode
tables are, so all help is appreciated. :) Though I found this

  http://www.z80.info/z80undoc.htm

... and can try to use it to enhance WLA DX's Z80 support...

vhelin, 2017-01-10:
Just made a commit to master that should add support for "sub ixh" and "sub ixl", and kept the old aliases for these.

PS. If you edit the opcodes_z80.c or other such files, please make sure the new entries are in suitable places in the array. By that I mean that the parser goes through them one by one, from first to last, and when it finds the first entry that could match, it'll use it.

closed 2017-01-10 by vhelin