vhelin/wla-dx #129: Allow specifying the system/region in .SMSTAG

maxim-zhao, 2016-10-26:
The high nibble of $7fff (or $1fff or $3fff) specifies the region and system:

Value | System/region
----------|-----------
$3	| SMS Japan
$4	| SMS Export
$5	| GG Japan
$6	| GG Export
$7	| GG International

It'd be nice to be able to specify it, right now it's always $4.

maxim-zhao, 2016-10-26:
It looks like it's actually filled in via .COMPUTESMSCHECKSUM leading to compute_sms_checksum(), which is slightly weird but perhaps necessary as the mem_ functions don't seem to allow readback? Possibly the solution is to extend .smstag to allow full control over the tag contents, including the checksum, product code, ... and I guess I ought to do the work...

nicklausw, 2016-11-17:
Maybe .SMSTAGHEADER would work? The sms's header is weird, there's some things I'd rather leave to the assembler to fill in about it so let's try not to add on to smstag just yet.

maxim-zhao, 2016-11-17:
The trouble is that the system/region is half of a byte where the other half is important for the checksum part of the header, and there doesn't seem to be a way to share bytes between directives like that. The least worst thing to do is to have one directive to control the whole (Sega) header, not the two we have today.
* TMR SEGA is required for SMS export and Game Gear
* Checksum is required for SMS export
* Region code is required for SMS export
* Other bytes might be nice to be able to control, e.g. for authentic "recompilation" with product codes and programmers' initials in the unused bytes
My use case was actually that I was compiling a [Game Gear game mod](https://github.com/maxim-zhao/mappy-gg-translation) and it had space in there for an SDSC header; but that then invokes .SMSHEADER which marks it as a Master System game, which is not great. There's no actual effect on emulators or real hardware, though.

nicklausw, 2016-11-17:
Why not have the directives use separate bytes, then wlalink combines them with an OR statement?
Like, if `a` is the high and `b` is the low, use `(a<<8) | b` to make the new byte. The linker's perfectly capable, though maybe the object format isn't.

vhelin, 2017-05-05:
Ok, I added .SMSHEADER that should (work and) fix this issue. Is it working for you guys?

maxim-zhao, 2017-05-05:
It looks good, I'll test it when I get time. 

One use of it is for disassemblies, where the goal is a 100% identical binary - so it may be desirable to control the "reserved bytes", since they do not have consistent values. I'd prefer to leave them unassigned (and thus set to .emptyfill) and then be able to fill them in with a .db if wanted.

vhelin, 2017-05-05:
> It looks good, I'll test it when I get time.
>
> One use of it is for disassemblies, where the goal is a 100% identical
> binary - so it may be desirable to control the "reserved bytes", since they
> do [...]

I think I'll add RESERVEDBYTES to .SMSHEADER to address this, I think
that's a more general solution to this issue?

maxim-zhao, 2017-05-15:
I just updated to latest and tested, and I have an annoying extra request - can I specify the ROM size nibble and consequent checksum range? Again, I'm going for a bit-identical disassembly, so inferring it from the actual size is wrong (it may not always match) and always using $c (for 32KB) was a decent hack at the time but if we're going to support all possibilities I guess we need:

```
checksumsize n ; either hex pass-through or a translation from a KB number?
checksum xxxx ; to force the checksum to some other value
```

I'll see if I can put together a PR...

vhelin, 2018-05-07:
Any news on the PR, or should I try to code this myself? :)

maxim-zhao, 2018-05-09:
No news at all, sorry. I've been away from WLA DX for a while, I'll probably circle back soon.

vhelin, 2022-02-06:
Everything mentioned in this issue should now be implemented so I'm closing this issue. Please reopen this / open a new issue if something is not working.

closed 2022-02-06 by vhelin

maxim-zhao, 2022-02-08:
Well, I very belatedly gave it a try and it works well for me. The last part (forcing the checksum to an invalid value) is still needed for a bit-identical result on some disassemblies, especially for Game Gear, but I can live without it.