vhelin/wla-dx #130: MAX_NAME_LENGTH can't be extended

maxim-zhao, 2016-12-08:
pass3.c seems to have a lot of this sort of thing:

		fscanf(f_in, "%64s ", tmp);

which frustrates attempts to exceed 64 chars. My quick fix was:

    #define XSTRINGIFY(x) #x
    #define STRINGIFY(x) XSTRINGIFY(x)
    #define STRING_READ_FORMAT ("%" STRINGIFY(MAX_NAME_LENGTH) "s ")

In this day and age, could we raise the default above 64? I managed to get there with this:

    _LABEL_AFA5_Emit3bppTileDataFromDecompressionBufferToVRAMAddressHL

nicklausw, 2016-12-08:
Flashbacks to reading in the docs about changing max length being a quick (and functional) fix. And that's why we need to test things...

I don't see the problem with just having the preprocessor handle it. Can C even support it being done any other way?

maxim-zhao, 2016-12-09:
The interwebs suggest you can alternatively ``sprintf()`` the string for ``fscanf()`` to use, but given that it's a constant, I don't see the need.

Also worth noting that the ``#define`` is in WLA DX ``defines.h`` and *also* in the WLALINK ``defines.h`` - but with a [different](https://github.com/vhelin/wla-dx/blob/master/wlalink/defines.h#L51) [value](https://github.com/vhelin/wla-dx/blob/master/defines.h#L8). I didn't try changing just one.

commit 26cbcbc6 (2017-01-07): Support reading in extended-length names, fixes #130
commit f309686e (2017-01-07): Merge pull request #134 from maxim-zhao/max_name_length

Support reading in extended-length names, fixes #130
closed 2017-01-07 by vhelin