vhelin/wla-dx #136: _sizeof_[label name]

maxim-zhao, 2017-02-07:
It's be nice if there was an auto-generated symbol for each named label which was the count in bytes from it to the next such label. This would make it a lot easier to capture the lengths of data blobs, for example:
```
label:
.asc "Some message here"
.define _sizeof_label next_label-label ; automate this
next_label:
```
Typically I find that the .define doesn't work nicely because of the lexical positioning typically being after where I want to use it anyway.

vhelin, 2017-05-16:
Ok, WLALINK now generates these definitions. I hope it works 100%. :)

maxim-zhao, 2017-05-16:
Awesome! I'll give it a spin later.

closed 2017-05-16 by vhelin

Ramsis-SNES, 2017-05-19:
This is indeed a cool feature, but unfortunately I can't get it to work. If I replace e.g. [this line](https://github.com/Ramsis-SNES/furryrpg/blob/master/furryrpg_text.inc.asm#L289):

`cmp	#(SRC_DiagPointerEng_END-SRC_DiagPointerEng)/2`

with this:

`cmp.w	_sizeof_SRC_DiagPointerEng/2`

then I get a value of zero instead of $BC. What am I doing wrong?

Also, it would be nice if the current size of the Accumulator would be acknowledged so `.w` could be omitted. :)

vhelin, 2017-05-20:
Can you post here the code? E.g. The section that contains both of the
labels.

In the case of _sizeof_ defines, the linker creates then, and that is too
late for size deduction.

Ramsis-SNES, 2017-05-20:
Sure, this is the file with the labels: https://github.com/Ramsis-SNES/furryrpg/blob/master/data/ptr_dialog_eng.inc.asm

The two labels are at the beginning and end of the file, respectively. (I didn't remove the `..._END` label when testing `_sizeof_`, BTW.)

For further reference, this is the main assembly file which contains most (if not all) `.INCLUDE`s: https://github.com/Ramsis-SNES/furryrpg/blob/master/furryrpg.asm

vhelin, 2017-05-20:
Thanks!

Looking at the first file I have no idea why the distance between the
labels is calculated incorrectly as the sources are as simple as it gets.
I'll try to debug this later today.

vhelin, 2017-05-20:
I wrote a similar testcase, but there the size of _sizeof_ definition was correct. If you make WLALINK to write out a symbol file, what's the value of _sizeof_SRC_DiagPointerEng there?

Ramsis-SNES, 2017-05-20:
Many `_sizeof_`s are 00:0000 (including both dialog pointer tables), but some appear to be correct (like `_sizeof_STR_SoftwareVersionNo` or `_sizeof_SRC_Palette_Portrait_Char1`):

```
00:0000 _sizeof_STR_Software_Title
00:0000 _sizeof_SRC_Palettes_Text
00:0000 _sizeof_GFX_Items_Eng
00:0000 _sizeof_SRC_DiagPointerEng ; dialog pointers
00:0000 _sizeof_SRC_DiagPointerGer ; dialog pointers
00:0000 _sizeof_SRC_HDMA_BG12CharData
00:0000 _sizeof_SRC_RamsisPresentsMap
00:0000 _sizeof_SRC_SoundEnginesMap
00:0000 _sizeof_GFX_WorldMap
00:0000 _sizeof_SRC_IoTmappal
00:0000 _sizeof_SRC_spc700_driver
00:0000 _sizeof_SRC_track_00_pointers
00:0000 _sizeof_SRC_track_01_pointers
00:0000 _sizeof_SRC_track_02_pointers
00:0000 _sizeof_SRC_track_04_pointers
00:0000 _sizeof_SRC_track_05_pointers
00:0000 _sizeof_SRC_track_06_pointers
00:0000 _sizeof_SRC_track_07_pointers
00:0000 _sizeof_SRC_track_08_pointers
00:0006 _sizeof_SRC_track_09_pointers
00:0000 _sizeof_SRC_track_00_samples
00:0000 _sizeof_SRC_track_01_samples
00:0000 _sizeof_SRC_track_02_samples
00:0000 _sizeof_SRC_track_04_samples
00:0000 _sizeof_SRC_track_05_samples
00:0000 _sizeof_SRC_track_06_samples
00:0000 _sizeof_SRC_track_07_samples
00:0000 _sizeof_SRC_track_08_samples
00:0014 _sizeof_SRC_track_09_samples
00:0002 _sizeof_STR_SoftwareVersion
00:0004 _sizeof_STR_SoftwareVersionNo
00:0002 _sizeof_SRC_Palettes_Items
00:0006 _sizeof_STR_SoftwareMaker
00:0020 _sizeof_SRC_Palette_Portrait_Char1
00:000f _sizeof_SRC_Palette_Portrait_Char2
...
```

vhelin, 2017-05-20:
I think I found the problem: if a section ends to a label, then that label
gets the same address as the next section's first label, if the next
section begins with a label, thus the distance between two consecutive
labels is zero. I hope to fix this later today...

vhelin, 2017-05-20:
Ok, I made it so that section's last label doesn't get _sizeof_, and also _sizeof_'s don't cross section boundaries. Should work now better.

Ramsis-SNES, 2017-05-20:
Yay, seems to work just fine, if I keep the modulo sign (`#`) in order to force immediate addressing mode, that is. :D

maxim-zhao, 2017-06-02:
Bug: it generates duplicate labels for "section-local" labels (starting with '_'). My solution:

```
    /* put the label into the global namespace */
    if (try_put_label(global_unique_label_map, l) == FAILED) {
      free(labels);
      return FAILED;
    }
````
->
````
    /* put the label into the relevant namespace */
    struct map_t* label_map = labels[j]->section_struct == NULL ? global_unique_label_map : labels[j]->section_struct->label_map;
    if (try_put_label(label_map, l) == FAILED) {
        free(labels);
        return FAILED;
    }
```
The variable needs to be hoisted upwards to be ANSI C...

vhelin, 2017-06-03:
Thanks!

I tried this, but then I get

main.o:main.s:42: FIX_REFERENCES: Reference to an unknown label
"_sizeof_label".

for my test case... According to debug output _sizeof_label is generated,
but fix_references() doesn't seem to find it. Could you try to fix this and
generate a pull request?

maxim-zhao, 2017-06-05:
I've iterated a few times since this, I'll try to extricate it into a PR. The gist of it is that the label should go into the section namespace if the base label is a _local one, the global one otherwise. I also made it assign a _sizeof_label for the last label in a section, since often I'll have one section per ``.incbin`` to allow better packing.

vhelin, 2017-06-05:
Sounds good! I just started to think that do we need _sizeof_ for local
labels? Why not just ignore them?

maxim-zhao, 2017-06-05:
I have a use case: a function in a section alongside its data blob, needing to know its blob's size, but that blob only has a local label pointing at it.

One other thing is to check what this does to "@" labels, I'm not sure if that makes sense yet.

maxim-zhao, 2017-06-06:
@ labels look OK, but I have another tricky case: ``.dstruct`` inserts a label, and that makes sense (although making it optional might be nice - sometimes I don't want that label), but it's logically the same as a user-entered label - but I think it shouldn't be. That may mean making a new label type which is stretching my comfort with the code.

One more: should _local labels interrupt non-local ones? So:

````
data:
.dw _data1, _data2
_data1: .incbin "data.bin"
_data2: .incbin  "data2.bin"
next_label:
````
Would you expect ``_sizeof_data`` to be 4 or everything up to ``next_label``? I'm indecisive, currently it is 4.

maxim-zhao, 2017-06-26:
One more: `.dstruct` also inserts labels for all the members of the struct, but then the first member is sorted before the struct itself (not sure why) and gets a size of 0, and the struct gets the size of its first member... overall, I think I want to discard all the `.dstructs` from the `_sizeof_` generation, but there doesn't seem to be a way to do this because as mentioned above, they appear identical to regular labels.

And another: logically, I suppose hierarchical labels (`@`-prefixed) ought not to terminate a non-`@` label's logical size. I'm not inclined to try too hard with that, though.

Drenn1, 2017-07-03:
FYI, "@" labels get prefixed with their parent name before anything is done with them. This means that there isn't much special-case code that needs to deal with them.

Anyway, this breaks namespaces a little bit. It makes it impossible to have a label with the same name in different namespaces. A simple fix could be just to add the _sizeof label to the same section? I'll look into it.

maxim-zhao, 2017-07-03:
The _sizeof_ definitions are using the prefixed names, so there's no issue there. My concern is that this:

````
label:
    nop
@1:
    nop
@@2:
    nop
label2:
````

...has _sizeof_label = 1, _sizeof_label@1 = 1, _sizeof_label@1@2 = 1 instead of _sizeof_label = 3, _sizeof_label@1 = 2, _sizeof_label@1@2 = 1 which would make more sense given the nesting. It is this which I did not feel like trying to solve.

commit 716f7170 (2017-05-20): Fixed #136 to work better.